dotintent/react-native-ble-plx #1: Unable to install module via npm install

Firaenix, 2016-08-20:
Hi,

I'm trying to add your library to my React Native application by following your guide on both Linux and Windows, to which I run into the same problem when `npm install` tries to run ./build_ios_frameworks.sh

I have attached the stack trace

**Ubuntu**
`╰─$ npm install

> react-native-ble-plx@0.1.3 install /mnt/f/Development/WetnessApp/node_modules/react-native-ble-plx
> ./build_ios_frameworks.sh

./build_ios_frameworks.sh: line 4: carthage: command not found
./build_ios_frameworks.sh: line 5: carthage: command not found
npm ERR! Linux 3.4.0+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v6.4.0
npm ERR! npm  v3.10.3
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! react-native-ble-plx@0.1.3 install: `./build_ios_frameworks.sh`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the react-native-ble-plx@0.1.3 install script './build_ios_frameworks.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the react-native-ble-plx package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./build_ios_frameworks.sh`

**Windows** 

`$ npm install

> react-native-ble-plx@0.1.3 install F:\Development\WetnessApp\node_modules\react-native-ble-plx
> ./build_ios_frameworks.sh

'.' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.4.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! react-native-ble-plx@0.1.3 install: `./build_ios_frameworks.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the react-native-ble-plx@0.1.3 install script './build_ios_frameworks.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the react-native-ble-plx package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./build_ios_frameworks.sh`

Cierpliwy, 2016-08-22:
Thanks for your report!

Please install 0.1.5 version of react-native-ble-plx which should check if your operating system contains carthage binary for building iOS frameworks. It should work at least on Ubuntu. On Windows bash may be necessary to go through installation process.

Firaenix, 2016-08-22:
Thanks very much for your help!

Cierpliwy, 2016-08-23:
Is it working properly now? If so, I'll close the issue.

Firaenix, 2016-08-23:
Yep all good, thanks!

closed 2016-08-23 by Cierpliwy

Cody-G-G, 2016-11-14:
@Cierpliwy am I understanding correctly that this means I cannot use react-native-ble-plx if I'm developing an iOS app using React Native on Windows (since I'd need Carthage)?