dotintent/react-native-ble-plx #2: Module 'BleClientManager' not found

patrickhoehn, 2016-08-31:
I get this error while trying to build your example in Xcode: `/examples/ReactBLEScanner/node_modules/react-native-ble-plx/ios/BleClient.m:10:9: fatal error: module 'BleClientManager' not found`

konradrodzik, 2016-08-31:
Did you done 'npm install' being in 'examples/ReactBLEScanner' directory?

patrickhoehn, 2016-08-31:
Yes, tried it three times with your version 0.1.5 and twice with 0.1.3. Did I skip anything else? I just downloaded the ZIP, navigated to 'examples/ReactBLEScanner', ran 'npm install', opened the xcode file and built it to run on my iPhone.

konradrodzik, 2016-08-31:
That should be it. Just checked 0.1.5 version and npm install did the whole work.
I was able to compile and run on my iphone.
Whats is your XCode version and could you past here the whole console output from npm install?

when you run npm install it will compile 'react-native-ble-plx' library.
We are using 'carthage' there, so it needs to be installed in your system.
Please follow instructions from
https://github.com/Carthage/Carthage

Let me know if that helps.

patrickhoehn, 2016-08-31:
That's it! Installing Carthage fixed the problem. Maybe you should mention in the readme that Carthage is required. Thank you!

closed 2016-08-31 by patrickhoehn

konradrodzik, 2016-08-31:
Great! We will add info about carthage. Thanks.

jcrjaci, 2016-09-17:
@konradrodzik im using ubuntu, to build an android react native app. I'm getting the same error.
Carthage isn't only for IOS?

konradrodzik, 2016-09-18:
@jcrjaci could you give us full log from your android problem?
Carthage is needed for iOS projects.

jcrjaci, 2016-09-18:
@konradrodzik problem solved.
Stupid error.
I missed the import
```
import com.polidea.reactnativeble.BlePackage;
```
in the mainApplication.java.

The import is missing in Read.me too.

konradrodzik, 2016-09-18:
Thanks, we will check it : )

jcrjaci, 2016-09-18:
@konradrodzik btw, nothing about the issue, but if you know to answer me i appreciate.
I only test the basics of the library.
like find dispositive and connect to them.

But now i need to send them hexadecimal commands, and read the answer.
 i see that i need to convert my commands to base64.
but this should work right?
i didn't test yet

Cierpliwy, 2016-09-19:
@jcrjaci We don't force users how they want to operate on values. You can add additional dependency: for example 'buffer' library which will let you just do that.

konradrodzik, 2016-09-19:
You can see in the example app how 'Buffer' can be used:
https://github.com/Polidea/react-native-ble-plx/blob/0eb39ce1c7f39fde5d5eafd4e39b6dad21c147e0/examples/ReactBLEScanner/app/characteristics/CharacteristicDetailsComponent.js

julienvincent, 2017-05-29:
I just ran into this issue and have two comments regarding this:

> Great! We will add info about carthage. Thanks.

I'm assuming @konradrodzik added the following line to the readme:

> Make sure that you have Carthage installed on your system.

This only implies that Carthage is required for _manual_ installs. If performing the automatic approach (via rnpm) there is no mention of the Carthage requirement.

I suggest that the readme change to reflect the following:

## Automatic

```
brew install carthage
npm install --save react-native-ble-plx
react-native link
```

Additionally, the npm|yarn install __succeeds__ even without carthage on the system. This should probably be changed to cause the install to fail if carthage is not installed locally.

farshidbakh, 2017-10-24:
It's very important that you first install the "Carthage" and then install the "react-native-ble-plx", otherwise you will see the same error massage "BleClientManager not found."