dotintent/react-native-ble-plx #5: Question - Motivation behind this library

ksachdeva, 2016-09-01:
Hi,

I am a big fan of your work on providing reactive extensions for BLE i.e. RxAndroidBLE & RxBluetoothKit and wondering what was the motivation to build the react native plugin/part that wraps the Rx libraries and end up exposing the Promise interface to javascript.

There exists https://github.com/innoveit/react-native-ble-manager that achieves the same objective of exposing the BLE stack so am wondering since it is not the javascript observable interface that is exposed here what is the reason/motivation to create this library.

Regards & thanks
Kapil

Cierpliwy, 2016-09-02:
Hello,

Let me start with mentioned the library, as it implements a very similar API currently. In our opinion writing iOS/Android native libraries from scratch is not a good thing. Especially with Android, which has a lot of quirks due to different BLE stack implementations of each vendor. We wanted platform specific issues to be solved on the native side.

RxAndroidBLE and RxBluetoothKit are using ReactiveX, because of easy compatibility and very centralized error handling (which is very important) as we don't want to ignore even the slightest reported error. That's why these libraries are used by react-native one and we feel more confident that our code works properly on each device. Basically, we don't want to double the effort.

However, API suggested for react-native-ble-plx is still in its development stage as I'm not sure if I like it or not. We considered using RxJS, but I was worried about performance issues, additional dependency and familiarity with a framework required for potential users. There are also limitations from react-native bridging side which are very annoying and only make it more difficult to pass streams of data between JS/Obj-C/Java - we have to use events for that. Also there is direct support for promises.

The API is not settled as we didn't hit 1.0. If you have any suggestions, please share them with us. It's always good to hear your opinion.

Best regards,
Przemek

ksachdeva, 2016-09-02:
Hi Przemek,

Thanks for the reasoning. It makes sense. 

Regards
Kapil

closed 2016-09-02 by ksachdeva