dotintent/react-native-ble-plx #6: macOS

mane87, 2016-09-19:
Hi,
since this library works on iOS and the macOS Bluetooth framework should be the same (afaik), would it be possible to use this library together with react native for mac (1)

 #Thanks!

1) https://github.com/ptmt/react-native-macos

Cierpliwy, 2016-09-20:
Hello,

That's correct. This library is based on RxBluetoothKit, which already supports macOS. However I'm not sure how to resolve building difficulties on iOS/macOS systems. As you probably know during npm install process custom shell scripts is executed which currently builds iOS only frameworks. Unfortunately this process is quite long with carthage and I'm not sure if I want to delay it even more with macOS libs.

Ideally we could download prebuilt versions of frameworks, however ABI for Swift is not stable yet. 
I would appreciate any suggestions.

mane87, 2016-09-21:
Hi,
tbh honest i am relatively new to react native, so i have still a lot to learn. I am in the stage of evaluating react native for a future project. Being able to use this lib also on mac would be a hue plus for us.

Does this build only happens upon installation or with every compilation of the app?
Thanks!

Cierpliwy, 2016-09-21:
Quick answer is that there is nothing in current codebase which will not allow using this library with macOS. During `npm install` process iOS frameworks are built by default and they need to be rebuild each time `react-native-ble-plx` library is updated, to be sure that JS API and ObjC/Swift APIs are in sync.

Currently to support macOS after installation you can go to `node_modules/react-native-ble-plx/` edit `build_io_frameworks.sh` and change platform to `Mac`. Relaunch this script and as a result in folder `./ios/BleClientManager/Carthage/Build/Mac/` there will be macOS frameworks. Last step would be to change paths in Xcode project from `iOS` to `Mac` which are mentioned in `README.md` file.

However this procedure is not user friendly and the question is how user should select version of frameworks. We could build iOS and Mac together but it will extend setup time which can be annoying when using CI for building. On the other hand we can force user to launch custom scripts after `npm install` but he or she needs to remember to do it after every library update to not break things.

closed 2016-10-10 by literator

ddgromit, 2017-09-07:
For anyone trying to get this package working wth react-native-macos, here's what I learned so far:

- The Base SDK needs to be changed in both `BleClientManager.xcodeproj` and `BleClient.xcodeproj` to target "Latest macOS".  Otherwise `./build_io_frameworks.sh` will error with `Dependency "BleClientManager" has no shared framework schemes for any of the platforms: Mac`
- Comment out references to `RestoredState` in the `BleClientManager` project.  RxBluetoothKit [only exports](https://github.com/Polidea/RxBluetoothKit/blob/552cf7635c0563eb3eb145452a681167db7f049e/Source/RestoredState.swift#L29-L30) this type and the associated methods when building for iOS, and you will get undefined type errors otherwise.
- "Header Search Paths" in BleClient.xcodeproj needs to be modified to target react-native-macos instead of react-native.  e.g. `$(SRCROOT)/../../react-native-macos/React`

I called it quits when getting 'React/RCTDefines.h' errors when trying to build BleClient.  Interested to hear if anyone gets it working.