dotintent/react-native-ble-plx #7: Bluetooth is not supported

littletower, 2016-10-09:
Hi,

I'm on the latest macOs 10.12 (on a 2015 MacBook Pro), with an updated Xcode. When I run the provided example in Xcode, the app seams to start just fine, but when I click on the `Start scanning` button I get an error message `Bluetooth is not supported`.

What causes this error? Am I missing something here? Bluetooth is activated.

Thanks!

Kacper20, 2016-10-10:
It appears that support for CoreBluetooth is dropped in simulator, so the message is proper. 
This message is internally caused by `CBCentralManager` state being `Unsupported`, so we can't do anything with it. 
More info on this: [HERE](http://stackoverflow.com/questions/10024608/does-the-iphone-simulator-in-xcode-support-bluetooth-low-energy)

Cierpliwy, 2016-10-19:
As Kacper said, this issue is not related to our library. For now I'm closing this issue.

closed 2016-10-19 by Cierpliwy