dotintent/react-native-ble-plx #8: device.cancelConnection bug (and fix)

leo3linbeck, 2016-10-10:
First, thanks so much for this VERY helpful module. I'm just getting into it, but it's well engineered from what I see so far. Keep up the great work!

I ran across a bug when using the device.cancelConnection() helper function. I kept getting an error "options - Variable not found" or some such. I went into your code, and found this statement (lines 14-16 of src/Device.js):

`        this.cancelConnection = () => {
            return manager.cancelDeviceConnection(this.uuid, options)
        }
`

This looks like a copy-paste of the device.connect() helper function above. I changed these lines to:

`        this.cancelConnection = () => {
            return manager.cancelDeviceConnection(this.uuid)
        }
`

and the problem went away. Just thought you'd want to know. Again, thanks, and all the best. Cheers,
L3

literator, 2016-10-10:
@leo3linbeck thanks for reporting this. @Cierpliwy (main maintainer) is unavailable at the moment, but we'll try to resolve this issue quickly.

Cierpliwy, 2016-10-10:
It was clearly a bug. Thank your for your pull request. If you have any suggestions regarding API feel free to open another issue. We need more feedback before stabilization.

closed 2016-10-10 by Cierpliwy