dotintent/react-native-ble-plx #10: Can´t read from every characteristic

BigPun86, 2016-10-18:
Hey there, i am trying to read some characteristics from my BLE-Module -> RN4020 which controls a chip over UART.

Now your module works great until i try to read some specific characteristics....

error code: 700
"BleGattException{status=2, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}"

and sometimes 

error code: 600
"BleGattCannotStartException{status=-1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}"

Any ideas on this?

And a general question, is it possible to just send something to the connected ble device? Something like write(identifier, "P0") for example or does it have to be written to an characteristic?

Good work though :P

Cierpliwy, 2016-10-18:
It looks like you try to read characteristic which is not readable according to these status codes:
https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h. In the future I plan to make errors more unified and descriptive between platforms.

By verifying `characteristic.isReadable` value you can be sure if operation is allowed. Same rule applies to writes and notifications. 

BLE devices store their values in characteristics only. You can read them directly by calling `readCharacteristicForDevice` and filling all required UUIDs.

I hope it answers your question :)

BigPun86, 2016-10-19:
Okay that sounds really good. The thing is that i checked the readable attribute on that service. This service is for MLDP. I might need to activate this service somehow, but didn´t find out yet how. Do u have any kind of experience with MLDP Services? 

When i try to read MLDP Service i get following error:
```
BleGattException{status=2, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'
```

Cierpliwy, 2016-10-19:
Unfortunately I don't have any experience with RN4020. However I would suggest to use standard BLE Scanner on Android and check if this issue is for sure reproducible and not related to react-native-ble-plx library.

BigPun86, 2016-10-19:
Well actually thats my fault. I tried several libraries and i get the same errors there too. But i kept working with rn-ble-plx because it seems to be the cleanest. Thanks though :)

Cierpliwy, 2016-10-19:
As I can't help you with this issue I'm closing it. Thank you for your feedback :)

closed 2016-10-19 by Cierpliwy

vschmidt, 2021-04-12:
I have the same problem with ESP32. However, when I use the "nRF Connect" or "Scanner BLE" applications, I can read the values ​​without problems. Is there a parameter that makes a reading without validating with the ".isNotifiable" method?

Scanner BLE:
[image]

nRF Connect:
[image]