dotintent/react-native-ble-plx #12: Allow to write to and read from characteristic

Cierpliwy, 2016-10-19:
Implement following operations related to descriptors:

- [ ] Discover descriptors for characteristics
- [ ] Read descriptor value
- [ ] Write descriptor value
- [ ] Note in documentation that these operations cannot be used on Client Characteristic Configuration descriptor (on iOS) or implement a workaround.
- [ ] Add constants for predefined descriptor UUIDs

seanadkinson, 2017-03-28:
Was looking for functionality to configure enabling indicators vs notifications, and stumbled across this enhancement.  Am I correct that we'd need the ability to write to the client config descriptor in order to specify indicator instead of notification?  Do all subscriptions default to notification currently?

I see your note that this doesn't work in IOS.  Do they just not support changing the notification form or something?

Thanks for this great library, btw.  I'm still learning how to use it with our bluetooth device, so forgive me if any of my questions are naive.

Cierpliwy, 2017-04-03:
Hello there. On iOS there is a function called setNotifyValue(_:for:) which is responsible for enabling and disabling notifications/indications: https://developer.apple.com/reference/corebluetooth/cbperipheral/1518949-setnotifyvalue.

The most interesting part is: "If the specified characteristic is configured to allow both notifications and indications, calling this method enables notifications only."

So it means that you cannot enable indications when both options are available. We could write specific values directly to Client Characteristic Configuration descriptor with writeValue(_:for:):

But, again:
"You cannot use this method to write the value of a client configuration descriptor (represented by the CBUUIDClient​Characteristic​Configuration​String constant), which describes how notification or indications are configured for a characteristic’s value with respect to a client. If you want to manage notifications or indications for a characteristic’s value, you must use the set​Notify​Value(_:​for:​) method instead."

Therefore I decided to mimic iOS behaviour on Android platform. I'm not sure if anything can be done here. If you want to use indications only, I think it's the best way to disable notifications on firmware side.

seanadkinson, 2017-04-03:
Thanks very much for the info @Cierpliwy

closed 2019-03-23 by stale[bot]