dotintent/react-native-ble-plx #14: Support latest versions of backend libraries

Cierpliwy, 2016-10-19:
- [ ] Make sure to update RxBluetoothKit and RxAndroidBle to latest stable versions before release
- [ ] Make sure to all required features are implemented such as fetching global BLE state.

seanadkinson, 2017-03-15:
Hello, thanks for the nice library.

Do you know if the latest versions would fix this warning in xcode?

> CoreBluetooth] API MISUSE: <CBCentralManager: 0x171c7e180> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported

More info:

http://stackoverflow.com/questions/20956880/corebluetoothwarning-has-no-restore-identifier-but-the-delegate-implements

If not, I can fill out a separate issue if you like.  Thanks again

Kacper20, 2017-03-17:
Hello @seanadkinson 
You should not be worried about this. This is caused because our proxy object implements all of the `CBCentralManagerDelegate` methods including `willRestoreState`. Since you're not using this feature(this requires with passing identifier to initialiser) there is nothing to worry.

Unfortunately there is no way to not implement this method in a clear way, so warning is unevitable

Kacper

Cierpliwy, 2017-11-20:
Closing this as updating libraries is not a task which can be resolved. We will update native dependencies as soon as they will bring value or fix bugs.

closed 2017-11-20 by Cierpliwy