dotintent/react-native-ble-plx #22: Monitoring characteristic on multiple devices

leo3linbeck, 2016-10-25:
I am trying to monitor a characteristic on multiple devices. I call something like:

```
remove_functions{device_uuid+':'+char_uuid} = ble.monitorCharacteristicForDevice(device_uuid, service_uuid, char_uuid,
        (error, characteristic) => {
            if (error) {
                console.log('ERROR ', error);
            }
            else {
                const value = new Buffer(characteristic.value, 'base64').toString('utf8');
                console.log('listener called', value)
            }
        }
    );
```

I call this twice, once for each device. The call is successful, no errors, and I can see the "remove" functions in the remove_functions object.

When a characteristic changes on one device, the monitoring works great - I can see the changes on the console, as events are properly emitted. However, when I change the characteristics on the other device, the listener is never invoked.

I am able to read a characteristics on multiple devices, and I am able to monitor several characteristics on the same device (I've tested that, and it works great) but I can't seem to monitor multiple devices.

Any help or suggestions where to look would be greatly appreciated. I did a cursory look at the javascript library, and nothing jumped out at me.

Thanks again for this library - it's been a big help!

Cheers,
L3

Cierpliwy, 2016-10-25:
Thank you for your report. I'll try to reproduce it and fix as soon as possible.

Cierpliwy, 2016-10-26:
I've rewritten notification implementation on notifications_fix branch, which should
fix your problem. If you can, please checkout new version directly from Github's branch:

Add to your npm dependencies:
`"react-native-ble-plx": "git://github.com/Polidea/react-native-ble-plx.git#notifications_fix"`

If everything is OK, we will merge it to master.

leo3linbeck, 2016-10-26:
Tried, but couldn't build. Got a ton of Swift warnings, and a few Swift errors that killed the build.

Here's where things started to go awry:

`*** Checking out RxSwift at "2.6.0"
*** Checking out RxBluetoothKit at "2.0.1"
*** xcodebuild output can be found in /var/folders/69/gg1tw4vs23b097n605f62nv00000gn/T/carthage-xcodebuild.9B44En.log
*** Building scheme "RxSwift-iOS" in Rx.xcworkspace
--- xcodebuild: WARNING: Unable to open project file '/Users/leo3/github/local/brevitest/app/development/brevitest_v2/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.playground' in workspace '/Users/leo3/github/local/brevitest/app/development/brevitest_v2/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.xcworkspace'.
warning: target specifies SWIFT_VERSION = '2.3', but it is overridden by TOOLCHAINS = 'com.apple.dt.toolchain.XcodeDefault'
warning: target specifies SWIFT_VERSION = '2.3', but it is overridden by TOOLCHAINS = 'com.apple.dt.toolchain.XcodeDefault'
/Users/leo3/github/local/brevitest/app/development/brevitest_v2/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/RxSwift/Observables/Observable+Creation.swift:22:5: warning: 'warn_unused_result' attribute behavior is now the default
`

There were then lots of warnings, and then two errors that killed the compile:

`/Users/leo3/github/local/brevitest/app/development/brevitest_v2/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/RxSwift/Rx.swift:26:32: error: 'inout' before a parameter name is not allowed, place it before the parameter type instead
/Users/leo3/github/local/brevitest/app/development/brevitest_v2/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/RxSwift/Rx.swift:35:32: error: 'inout' before a parameter name is not allowed, place it before the parameter type instead
`

In another file, there were more errors, and they looked like this:

`/Users/leo3/github/local/brevitest/app/development/brevitest_v2/node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/RxSwift/Schedulers/RecursiveScheduler.swift:15:26: error: function types cannot have argument label 'state'; use '_' instead
`
The final statement was:

`The following build commands failed:
	CompileSwift normal arm64
	CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler
(2 failures)
`

Anyway, looks like I need to do something with Xcode compiler settings to make this work, but don't know what. I did a complete clean (and clean folders) on Xcode, zapped node_modules, and started from scratch, but same outcome.

Any ideas? Cheers,

L3

Cierpliwy, 2016-10-26:
It is strange. I don't have any issues with building...
I'm currently using Xcode Version 8.0 (8A218a) and Carthage 0.18.1.
Are you able to build origin/master branch? 

It looks like your frameworks are building for Swift 3.0, not 2.3. I guess
it's an issue with old carthage version probably. Make sure to update to the latest one.

leo3linbeck, 2016-10-26:
Haha - don't you love dependency-based development? ;-)

Alas, I'm using the same Xcode version and same Carthage version. I will dig further and report back. Cheers,

L3

leo3linbeck, 2016-10-26:
Found the issue.

The Cartfile has the following:

`Github "Polidea/RxBluetoothKit" == 2.0.1`

I think this should be:

`Github "Polidea/RxBluetoothKit" == 3.0.5`

or something like that. When I make that change, it builds nicely. Maybe you've got some environment variable set up that avoid this issue, but the versioning struck me down.

In an abundance of caution, I also changed Cartfile.resolved to:

```
github "ReactiveX/RxSwift" "3.0.0"
github "Polidea/RxBluetoothKit" "3.0.5"
```

Now onto testing...

Cheers,
L3

leo3linbeck, 2016-10-26:
Well, I can't get the build to work. Changing from 2.3 to 3.0 as described above removed a lot of the problems, but I still get the following error messages:

```
--- xcodebuild: WARNING: Unable to open project file '/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.playground' in workspace '/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.xcworkspace'.
*** Building scheme "RxTests-iOS" in Rx.xcworkspace
--- xcodebuild: WARNING: Unable to open project file '/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.playground' in workspace '/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.xcworkspace'.
--- xcodebuild: WARNING: Unable to open project file '/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.playground' in workspace '/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxSwift/Rx.xcworkspace'.
*** Building scheme "RxBluetoothKit iOS" in RxBluetoothKit.xcodeproj
*** Building scheme "BleClientManager" in BleClientManager.xcodeproj
** BUILD FAILED **

The following build commands failed:
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Utils/DisposableMap.swift
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Utils/SafePromise.swift
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleEvent.swift
	CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler
(5 failures)
ld: warning: directory not found for option '-F/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxBluetoothKit/Carthage/Build/Mac'
ld: warning: directory not found for option '-F/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxBluetoothKit/Carthage/Build/Mac'
ld: warning: directory not found for option '-F/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxBluetoothKit/Carthage/Build/Mac'
ld: warning: directory not found for option '-F/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxBluetoothKit/Carthage/Build/Mac'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleModule.swift:246:19: error: expected 'let' in conditional
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleUtils.swift:12:11: error: 'SequenceType' has been renamed to 'Sequence'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleError.swift:11:18: error: 'ErrorType' has been renamed to 'Error'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleError.swift:41:11: error: 'ErrorType' has been renamed to 'Error'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:19:44: error: value of type 'Data' has no member 'base64EncodedStringWithOptions'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:25:30: error: value of type 'Data' has no member 'base64EncodedStringWithOptions'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:30:14: error: cannot convert value of type '[String]?' to expected argument type 'AnyObject?'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:34:14: error: cannot convert value of type '[String]?' to expected argument type 'AnyObject?'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:38:14: error: cannot convert value of type '[String]?' to expected argument type 'AnyObject?'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Utils/SafePromise.swift:11:28: error: single argument function types require parentheses
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:43:21: error: use of unresolved identifier 'RSSI'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:59:21: error: value of type 'UUID' has no member 'UUIDString'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:77:21: error: type 'UUID' has no member 'fullUUIDString'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:87:21: error: type 'UUID' has no member 'fullUUIDString'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:104:15: error: enum case 'Unknown' not found in type 'BluetoothState'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:105:15: error: enum case 'Resetting' not found in type 'BluetoothState'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:106:15: error: enum case 'Unsupported' not found in type 'BluetoothState'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:107:15: error: enum case 'Unauthorized' not found in type 'BluetoothState'
/Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift:108:15: error: enum case 'PoweredOff' not found in type 'BluetoothState'
[... cut ...]
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleExtensions.swift
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Utils/DisposableMap.swift
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/Utils/SafePromise.swift
	CompileSwift normal arm64 /Users/leo3/github/local/react-native-ble-plx/ios/BleClientManager/BleEvent.swift
	CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler
(5 failures)

```

Any ideas? Thanks in advance. Cheers,

L3

leo3linbeck, 2016-10-27:
SUCCESS!

Sorry for the long and winding road, but I was able to do the build by telling Xcode to use Swift 2.3 (this can be done by Edit -> Convert -> To Current Swift Syntax... and then selecting v2.3.

Anyway, the REALLY good news is that your changes work - I'm getting notifications from two devices (vs. just one before). So while I haven't extensively tested this, I would give it the thumbs up.

I will continue to use it and the rest of the library, and if I discover any more issues I will dutifully report them. Thanks again for the great work! Cheers,

L3

Kacper20, 2016-10-27:
Hi,
To clarify things a bit -
While doing conversion iOS is switching magic flag in Build Settings which you can see as "Use Legacy Swift version". YES tells xcode to build with Swift 2.3, NO builds with Swift 3.0. Since this library supports Swift 2.3 and depends on RxBluetoothKit 2.X which also supports Swift 2.3 essential part is setting the flag to YES(and this is what convert to Swift 2.3 has done under the hood). 
RxBluetoothKit 3.X supports Swift 3.0 and is not currently compatible with this lib.

Cheers

Cierpliwy, 2016-10-27:
I've updated build script which forces usage of Swift 2.3 when building dependencies. Could you verify your build, previously cleaning all dependencies, @leo3linbeck ?

leo3linbeck, 2016-10-28:
@Cierpliwy,

Done. It builds beautifully. Thanks for your responsiveness! Cheers,

L3

Cierpliwy, 2016-10-28:
That's good news. Thank you for your help and patience! I'm closing this issue as it was merged to master branch and released as version 0.3.2.

closed 2016-10-28 by Cierpliwy