dotintent/react-native-ble-plx #25: Android setup information doesn't include importing BlePackage

Nealo, 2016-10-31:
Was following the directions for setting up for Android and for adding `new BlePackage()` into the `MainApplication.java` file, I get an error of "cannot resolve symbol 'BlePackage.' I tried copying the import from the example "`import com.polidea.reactnativeble.BlePackage;`" but that doesn't work either.

Cierpliwy, 2016-10-31:
Hello,

Just created new React Native project and configured it with updated steps - I've added more context info and there was missing min SDK configuration. After all applied changes I run `react-native run-android` and it built just fine.

Could you copy and paste errors from above command?

Nealo, 2016-10-31:
Okay thanks for updating that. It worked eventually but required numerous steps to get there. I put in the import line, and hit Run despite the errors. It came back saying I was missing `Build Tools Revision 23.0.3`. So I installed that. 

I then had to close Android Studio after updating, load it up again. At that point it created the `build.gradle` for `react-native-ble-plx`, which then of course let me import it. 

It's running now, thanks for the fast response.

closed 2016-10-31 by Nealo