dotintent/react-native-ble-plx #26: Build Failure with iOS

Nealo, 2016-10-31:
So I followed the tutorials for setting up both on iOS and Android.

Now that I've done that, iOS fails building each time with this error in Xcode:

`Check dependencies
Argument list too long: recursive header expansion failed at /Users/nealogrady/Dropbox/Sites/Sites/Bloom Dental/BloomDental/ios/../node_modules/react-native-ble-plx/android/build/tmp/expandedArchives/internal_impl-23.1.1.jar_7m76px62sfm8yakpu97l6umuv/android/support/v4/graphics.
`

When run through the terminal with `react-native run-ios`, it fails, takes forever but does eventually launch:

`
Check dependencies
Argument list too long: recursive header expansion failed at /Users/nealogrady/Dropbox/Sites/Sites/Bloom Dental/BloomDental/ios/../node_modules/react-native-ble-plx/android/build/tmp/expandedArchives/internal_impl-23.1.1.jar_7m76px62sfm8yakpu97l6umuv/android/support/v4/graphics.

** BUILD FAILED **

The following build commands failed:
	Check dependencies
(1 failure)
Installing build/Build/Products/Debug-iphonesimulator/BloomDental.app
`

In Xcode, however, it will not build at all.

Nealo, 2016-10-31:
I'm running the most recent versions of iOS, OSX, and Xcode.

Also `Embedded Content Contains Swift Code` is no longer in `Build Options` in Xcode from what I can see. Maybe it was replaced by `Always Embed Swift Standard Libraries`?

Cierpliwy, 2016-11-02:
Sorry for late response. I've updated installation steps for Xcode 8 IDE, as it will be used in future version of library. They are already on main Github's page. 

Make sure to remove previous recursive path and set:
In `Build Settings/Search Paths/Framework` search paths add path: `$(SRCROOT)/../node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS`

Also as you spotted, set:
`Always Embed Swift Standard Libraries` to `Yes`

Nealo, 2016-11-02:
Okay great, thanks for this!

But just to totally confirm: So there's no longer a Header Search Path, and I replace the old Framework Search Path with this one? And is this a non-recursive or recursive path?

Cierpliwy, 2016-11-02:
Header Search Path is there by default in new React Native projects. New path doesn't need to be recursive. Also in new 0.3.3 build there is no need to add RxCocoa dependecy anymore in Carthage copy-frameworks script.

Nealo, 2016-11-02:
Any idea why I might be getting this error now?

[image]

Nealo, 2016-11-03:
Sorry for the confusion there. I accidentally hit close while I was having some weird issues that took a while to resolve. It all seems to work now though. Thanks again.

closed 2016-11-03 by Nealo