dotintent/react-native-ble-plx #28: Listener string return value truncated

leo3linbeck, 2016-11-03:
Another possible bug:

I have a 24-byte UUID that I store in a characteristic. I setup a listener to monitor for changes in that value using the monitorCharacteristicForDevice function. I also read that characteristic value directly from time to time using the readCharacteristicForDevice function.

When I read the value directly and convert from Base64 to utf8, I get all 24 bytes.
When I get the value initially from the listener and convert, I get all 24 bytes.
However, in subsequent calls from the listener, I only get 20 bytes.

Here is some example output:

When reading directly:
```
Base64: MEQ0OUFEOUYxNDc4MTMyNjY1NTQwNTk5
utf8: 0D49AD9F1478132665540599
```
Initial response from listener:
```
Base64: MEQ0OUFEOUYxNDc4MTMyNjY1NTQwNTk5
utf8: 0D49AD9F1478132665540599
```
Subsequent responses from listener:
```
Base64: MEQ0OUFEOUYxNDc4MTMyNjY1NTQ=
utf8: 0D49AD9F147813266554
```

This one has me stumped. Any ideas?

One more thing I've noticed that may or may not be relevant: when setting up a listener on a characteristic, it does not get called initially (which is probably correct behavior). However, when I read the characteristic after setting up the listener, the listener gets called. Now this is not really expected behavior to me; seems like a call to read a characteristic that is Read+Notify should not trigger a listener event. As it is, when I set up a listener and do a read to get the initial state, I get two identical results - one from the read, and one from the listener. (This is what I meant above as the "initial response from listener").

Thanks in advance! Cheers,

L3

Cierpliwy, 2016-11-03:
Hello,

According to Bluetooth Specification Vol 3, chapter 3.4.7.1, maximum notification payload is defined as `ATT_MTU - 3` and and any bigger payload will be truncated. On most devices, from our experience `ATT_MTU` is negotiated as `23` bytes, which leave us with max `20` bytes of data possible to transfer by notification. Reads and writes are handled by chunks, that's why you can send or read `> 20B` by using them. Related post:
https://devzone.nordicsemi.com/question/84021/how-do-i-get-whole-31-byte-ble-notification-data-at-once/

Responding to your second issue: unfortunately on iOS devices there is only one delegate callback called: `peripheral(_:didUpdateValueFor:error:)` which is used at the same time for reads, notifications and indications. When value is received, we don't have enough context to determine if it was caused by notification or write. That's why after setting up listener and executing read, you will receive the same value on both listeners.

There are few things we can do about each case. First of all, when we receive notification which is smaller then expected, we should execute read to get full payload. Additionally to have better chance to distinguish notification/read values we can filter out expected events from monitor listener as we know exactly how many times we read characteristic.

I hope it helps you,
Przemek

leo3linbeck, 2016-11-03:
Thanks! That's super helpful - I didn't know about the Bluetooth spec. Makes perfect sense. I can easily deal with this - I'll just have a simple, quick notification characteristic that let's me know which characteristic changed, and then execute a read to grab the latest value. That should also deal with the second problem as well.

Thanks again for the quick response. Cheers,

L3

Cierpliwy, 2016-11-04:
I'll consider changing value monitoring behaviour in the future. I wrote it down as a one of points in Breaking Changes (issue #21).

For now I'm closing this thread.

closed 2016-11-04 by Cierpliwy