dotintent/react-native-ble-plx #30: [iOS] Bug with discoverAllServicesAndCharacteristics

joshreeves, 2016-11-11:
I am implementing a simple component that scans and connects to a BLE device. The device I am trying to connect to is BLE test board. Here is the code I am using:

```
this.bleManager.connectToDevice(device.uuid)
      .then((device) => {
        console.log("DISCOVERING");
        return device.discoverAllServicesAndCharacteristics();
      })
      .then((device) => {
        console.log("DISCOVERED");
        this.setState({
          connectedGogglesId: device.uuid,
          connectedGoggles: device
        });
      })
      .then(() => {
        console.log("SUCCESS");
      }, (error) => {
        console.log(error);
      });
```

On Android, everything works fine, all services and characteristics are discovered and the "SUCCESS" message is printed. One iOS, however, the connection is successful, I see the bluetooth icon flash and turn white, it prints the "DISCOVERING" log message, but it just hangs on the `discoverAllServicesAndCharacteristics` call and never completes the discovery.

I'm using `react-native 0.37` and the `master` branch of `react-native-ble-plx`. Any help or ideas at what might be causing this is GREATLY appreciated. Thanks!

joshreeves, 2016-11-11:
The test board I am connecting to is this one: http://www.cypress.com/documentation/development-kitsboards/cy8ckit-042-ble-bluetooth-low-energy-ble-pioneer-kit

Again, both Android & iOS can connect. But ONLY Android is able to receive a successful response from `discoverAllServicesAndCharacteristics`.

Cierpliwy, 2016-11-14:
Hello,

I tested library from master/origin branch with the same RN version and I couldn't reproduce your issues... Could you try Sniffator application from App Store (which uses react-native-ble-plx) and see if this problem still occurs after connecting to device?

Cierpliwy, 2017-06-02:
Any news on that?

jchirschy, 2017-08-25:
Hi all,

I'm facing a similar issue with IOS.
`discoverAllServicesAndCharacteristics` is rejecting with the following message : 
> Error: Device 2B71F4FC-AD01-45E0-ADB5-440174341B08 is not connected

I also get the following warning : 
> Sending `StateChangeEvent` with no listeners registered.

Nevertheless, when I test the connection with `.isDeviceConnected`, the promise resolves successfully. 
I'm using an Ipad Mini 4 and `react-native-ble-plx 0.6.1`.
Sniffator discovers characteristics properly, so as Lightblue.

Is there something new with this issue ?

Thanks for your help.

jchirschy, 2017-09-01:
Hi @Cierpliwy 

Is there something new with this issue or a workaround to setup ?

Thanks for your help
JeanClaude

pstanton, 2018-05-23:
I believe the message "Sending `StateChangeEvent` with no listeners registered." is caused by the module registering the desire to listen to the events in EventEmitter, before actually adding a listener. I also think it can be safely ignored, since I've tested and the bluetooth state change event is still notifying my listener, once added.

you can ignore the yellowbox message by adding:

```
import { YellowBox } from "react-native";
YellowBox.ignoreWarnings(["Sending `StateChangeEvent` with no listeners"]);
```

as documented [here](https://facebook.github.io/react-native/docs/debugging.html)

closed 2019-03-23 by stale[bot]