dotintent/react-native-ble-plx #31: [Android] Turn on Bluetooth

vikeri, 2016-11-14:
I don't know if this is in the scope of this library but it would be really nice to include in this library a function that enables Bluetooth. I know this only works for Android but it would still be useful.

Found some examples here: http://stackoverflow.com/a/20142972/1836121

Cierpliwy, 2016-11-14:
Thank you for a suggestion. I'll definitely add it to library.

stinodes, 2017-02-15:
@Cierpliwy any news on this feature?

Cierpliwy, 2017-02-15:
I don't have enough time right now to implement this feature. Also for 0.5 release I plan to focus on support for background mode, as it's very important for users of this library. I'll definitely welcome more contributions :)

stinodes, 2017-02-15:
Alright, that's totally understandable of course.  
If it becomes a requirement from my side, I might do a pull request, but for now it's just a nice to have and pretty low priority.

watadarkstar, 2017-05-12:
Did this ever make it in the code base? @Cierpliwy

arashd, 2017-06-20:
No updates on this feature?

watadarkstar, 2017-08-14:
I'm thinking of working on this issue on my free time! Will submit a pull request.

watadarkstar, 2017-09-14:
@saeedvaziry I started working on a PR but I still need to do some further testing. See: #138

darkrift, 2017-11-02:
This option is not available on iOS, however for iOS we can make it so that it redirects to the control panel to enable it as described here : 

https://stackoverflow.com/a/36129124

To unify it, we should probably prompt the user on Android. I'm not a fan of applications enabling stuff in my back without asking me if I disabled it.

brancooo1, 2018-03-07:
any news on this feature?

Cierpliwy, 2018-10-03:
@JasiuHimself Feature is implemented on the v1.0 branch. After proper testing, I plan to land it on the master.

JasiuHimself, 2018-10-04:
Amazing, can't wait!

brancooo1, 2018-10-05:
When is v1.0 expected to be public @Cierpliwy ?

Cierpliwy, 2018-10-05:
[It is public :)](https://www.npmjs.com/package/react-native-ble-plx)

closed 2018-10-09 by Cierpliwy

manoj-makkuboy, 2019-01-18:
How to turn on the bluetooth, I am not able to find any function to do so in documentation.

a1lu, 2019-01-25:
I found this (android only):
https://github.com/Polidea/react-native-ble-plx/blob/b4ccdf352f8ec06a718ef267ee7e31d7eabd87c6/src/BleManager.js#L225-L251

Cierpliwy, 2019-02-08:
@a1lu It's only for Android, because iOS doesn't provide any API to do the same thing.