dotintent/react-native-ble-plx #35: get the connected Bluetooth device informations

zhanguangao, 2016-11-21:
Can I get the connected Bluetooth device informations without any parameters ,and the device is  connected manually in system setting.

Cierpliwy, 2017-06-02:
If you mean normal Bluetooth devices I think it is currently impossible to do.

sikhote, 2017-07-10:
@Cierpliwy what about a BLE device that is paired within the system bluetooth settings? Possibly attain a list of paired devices?

Thanks for the great library.

Cierpliwy, 2017-11-29:
There is possibility to add functions:

- `function knownDevices(): Promise<Array<Device>>`
- `function connectedDevices(): Promise<Array<Device>>`

There are native counterparts on iOS side. On Android we could use internal caches for that purpose.

Cierpliwy, 2018-04-09:
```js
async devices(deviceIdentifiers: Array<DeviceId>): Promise<Array<Device>> 
```
and
```js
async connectedDevices(serviceUUIDs: Array<UUID>): Promise<Array<Device>>
```
were added in version 0.9

closed 2018-04-09 by Cierpliwy

bhenav, 2019-01-24:
Hi,

**Error**: knownDevices is not a function. 

Are you sure this is the best?
Can you write an example?

Thank you,
bhenav

askesis, 2019-01-25:
Hi, @Cierpliwy !

Is it possible to interact with paired (via OS, only android interesting for me) device through the package?

```js 
async devices(deviceIdentifiers: Array<DeviceId>): Promise<Array<Device>>
```
and other your example get required argument, but I don't know of devics UUID or something else...

Looks like I should know information about device earlier then use this package

fitbuff, 2020-08-18:
i also would like to interact with paired device (connected via Bluetooth System Settings).
is this possible with this package?