dotintent/react-native-ble-plx #36: issues installing

ErreMalote, 2016-11-25:
tall"
npm ERR! node v7.0.0
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE

npm ERR! react-native-ble-plx@0.3.3 postinstall: `bash build_ios_frameworks.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the react-native-ble-plx@0.3.3 postinstall script 'bash build_ios_frameworks.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the react-native-ble-plx package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     bash build_ios_frameworks.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs react-native-ble-plx
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls react-native-ble-plx
npm ERR! There is likely additional logging output above.

ErreMalote, 2016-11-25:
Solved by using Git BASH. Windows doesn't support .sh files on command line.

Cierpliwy, 2016-12-08:
I'll definitely check out and fix that issue on Windows.

albert-l-garrett, 2017-02-11:
I found this solution. 
http://www.windowscentral.com/how-install-bash-shell-command-line-windows-10

marwakhalil, 2017-05-05:
Darwin 15.3.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--save" "react-native-ble-plx"
npm ERR! node v7.9.0
npm ERR! npm  v4.2.0
npm ERR! code ELIFECYCLE
npm ERR! errno 1

npm ERR! react-native-ble-plx@0.4.0 postinstall: `bash build_ios_frameworks.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the react-native-ble-plx@0.4.0 postinstall script 'bash build_ios_frameworks.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the react-native-ble-plx package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     bash build_ios_frameworks.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs react-native-ble-plx
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls react-native-ble-plx
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/marwazribi/.npm/_logs/2017-05-05T14_59_43_032Z-debug.log

and i'm a mac user

Cierpliwy, 2017-06-02:
Can you check version 0.5.0 if this still happens?

BAVhel, 2017-07-10:
132426 verbose stack Error: react-native-ble-plx@0.6.0 postinstall: `bash build_ios_frameworks.sh`

Cierpliwy, 2017-07-12:
Is it your full error? Can you enter library directory and run this script by hand?:
```
cd ./node_modules/react-native-ble-plx/
./build_ios_frameworks.sh
```

BAVhel, 2017-07-12:
@Cierpliwy I ran the script on windows and I cant run sh without bash.

Cierpliwy, 2017-07-12:
Probably I will need to create node based script to be platform agnostic. If you want this feature faster, maybe you can tackle this problem and send me PR?

BAVhel, 2017-07-12:
@Cierpliwy I have the same error on examples/ReactBLEScanner

closed 2017-08-08 by Cierpliwy

jiazil, 2017-09-20:
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! react-native-ble-plx@0.6.1 postinstall: `bash build_ios_frameworks.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the react-native-ble-plx@0.6.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

jiazil, 2017-09-20:
xcode 9

PR #123 (2017-08-08): skip build_ios_frameworks.sh on windows
fixes #36 and #57