dotintent/react-native-ble-plx #37: Not able to read characteristics on iOS device

hnan, 2016-11-29:
I'm trying with this library with a very simple example. A button handler creates a BleManager object, and tries to connect to a device with a specified name, then tires to read a characteristic with a specified UUID.

It works on Android device. 

But on iOS device, it's able to scan and connect to the device, but not able to read any characteristics, in Chrome console, it shows the returned characteristics array is empty.

Cierpliwy, 2016-12-08:
Can you show me an example code with specific arguments values which you are passing and output which you are receiving?

LeonidVeremchuk, 2017-01-31:
Have this problem too.
```
connect() {
        console.log("Connecting to device :");
        console.log(this.device);
        this.bleManager.connectToDevice(this.device.uuid).then((device) => {
            this.bleManager.servicesForDevice(device.uuid).then((servicesForDevice) => {
                console.log("servicesForDevice");
                console.log(servicesForDevice);
            }).catch((error) => {
                console.log(error);
            });
        }).catch((error) => {
            console.log("connectiton error: " + error);
        });
    }
```

[logs:](https://i.gyazo.com/e1a24372ba3ab37b7ae4c4f45b775003.png)

Cierpliwy, 2017-01-31:
Did you call `discoverAllServicesAndCharacteristics()` on Device object successfully? Without it device won't have cached services and characteristics accessible.

LeonidVeremchuk, 2017-01-31:
I have updated my code and everything works fine now. Thank you very much!
```
    connect() {
        console.log("Connecting to device :");
        console.log(this.device);
        this.bleManager.connectToDevice(this.device.uuid).then((device) => {
            return device.discoverAllServicesAndCharacteristics();
        }).then((device) => {
            this.bleManager.servicesForDevice(device.uuid).then((servicesForDevice) => {
                console.log("servicesForDevice");
                console.log(servicesForDevice);
            }).catch((error) => {
                console.log(error);
            });

        }).catch((error) => {
            console.log("connect error: " + error);
        });
    }

```

Cierpliwy, 2017-01-31:
I'll make sure to improve documentation in upcoming releases.

closed 2017-01-31 by Cierpliwy