dotintent/react-native-ble-plx #38: Question: Background monitoring?

stami, 2016-12-07:
Hi!

Is it possible to monitor characteristics when the app is in background or sleeping?

Currently, without any additional setup, the app just does nothing when phone screen turned off. Monitoring (or receiving the values) continues after the app is opened again. (tested only on iOS)

Have you considered adding some kind of buffer to native code so that app could keep receiving data from the BLE device while screen is turned off? The buffer data would then be accessible from JavaScript.

Thanks for the great library!

vikeri, 2016-12-07:
I would really appreciate this feature as well

Kacper20, 2016-12-08:
Hi,
Have you turned on bluetooth background mode on iOS?
You could only do it entering proper info in plist. 
In this way, when bluetooth action is happening(for example notification on characteristic), your iOS app is woken up by the system and have a bit of time to do it's job.

Cierpliwy, 2016-12-08:
I certainly need to test it on both Android, iOS and check if any API changes or additional steps are required.

stami, 2016-12-13:
@Kacper20 Turning on `bluetooth-central` background mode doesn't seem to make any difference. Something more has to be done.. :)
The app continues from where it was left. Tested also with Instruments' Time Profiler and there seems to be no activity.

vikeri, 2016-12-13:
I'm working on this for Android. I think the app needs a scheduled background job to do this.

stami, 2017-01-19:
Hi guys!
I just found out that the background monitoring actually works out of the box (with background-mode turned on).

I had beed sending requests to my device to response me with data and that didn't work.
Instead of that, now I send the data from my device (Arduino with BLE chip) automatically without the need of the request and the app will continue to monitor and receive that data after turning display off. 😄 

This was tested on iPhone.

vikeri, 2017-01-19:
@stami Have you tried killing the app?

stami, 2017-02-13:
@vikeri Sorry for delay.. I hadn't notifications on for this thread.

Killing the app? In order to achieve what?
Currently the connection breaks if I reload the app. Haven't tried bundling and building release for some time...

vikeri, 2017-02-13:
@stami My experience is that it works as long as iOS/Android keeps the app in the memory. As soon as they kill the app due to inactivity or otherwise high memory load the BLE connection will be lost.

stami, 2017-02-14:
@vikeri I haven't got much knowledge on this but to me, that sounds something that cannot be helped even with native code.

vikeri, 2017-02-14:
@stami True, what I did was to put the code in a background job that schedules every 15 min. Then the app is woken up in the background and reconnects again to the Bluetooth. I did a library for it (only Android atm) [vikeri/react-native-background-job](https://github.com/vikeri/react-native-background-job)

mane87, 2017-02-21:
@Kacper20 Is there any known blocker for supporting state-restoration for iOS? Otherwise i would start working on that.

Kacper20, 2017-02-21:
@mane87 It's supported on backing `RxBluetoothKit` library, so I think that it's just the matter of creating API and wrapping this calls in JS.

@Cierpliwy any objections?

Cierpliwy, 2017-02-21:
I don't have any objections really. All I want is an API which won't require iOS/Android specific JS code. I'm not sure if it's doable though.

Kacper20, 2017-02-21:
To be precise:
I'm talking about `State Restoration` section from Bluetooth Programming Guide.
There's detailed description provided by Apple [here](https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1)

`RxBluetoothKit` has `listenOnRestoredState` method, which you should subscribe to after re-instantiating `BluetoothManager`(on `applicationDidFinishLaunching`). 
We need to have API from side of this library to wrap this one call, but I'm sharing @Cierpliwy doubts that this would have to be platform-specific logic, since this differs so much on Android and iOS.

So we would have to implement method that would have to be only available on iOS. Since I'm not really into ReactNative I don't know whether it's possible to do this in a nice way and how it'll affect library.

Does this answers your question @mane87?

mane87, 2017-02-21:
@Kacper20 @Cierpliwy Thanks for the fast answers guys. Sounds good. I'll have a look at it and see if i come up with a clean solution.

edencakir, 2017-03-14:
Hey guys, it's a really great feature, any news?

stami, 2017-03-14:
@edencakir Background monitoring is working.

As for state restoration, that is totally different question and should be discussed in another thread, in my opinion.

I opened this issue and consider it solved. Background monitoring was working, I wasn't just using it correctly as I was trying to poll the data instead of just monitoring changes.

I'm closing this issue now. Please create another one for state restoration. :)

closed 2017-03-14 by stami

edencakir, 2017-03-14:
Glad to hear that.

> Currently, without any additional setup, the app just does nothing when phone screen turned off. Monitoring (or receiving the values) continues after the app is opened again. (tested only on iOS)

Your comment made me confused, thank you for the clarification.