dotintent/react-native-ble-plx #39: Connect Promise Hangs

timscott, 2016-12-09:
I'm just getting started with this package and BLE.  I have the following code block:

```javascript
return manager.startDeviceScan(null, null, (error, device) => {
  if (device.name === LMU_NAME) {
    return device.isConnected().then(isConnected => {
      if (!isConnected) {
        return device.connect().then(response => {
          console.log(response);
        }).catch(err => {
          console.log(err);
        });
      }
    });
  }
});
```

A few seconds after ```connect``` is called, the phone prompts me to accept pairing, and I tap OK. Neither ```then``` nor ```catch``` is ever called.

Am I doing something wrong, or it's a bug?

Cierpliwy, 2017-11-29:
Closing as this is an overdue issue. Please feel free to reopen if you tested it with the latest version of a library.

closed 2017-11-29 by Cierpliwy

lukasz-wronski, 2019-10-28:
I got the same issue here. I am awaiting `connect` function inside of `try..catch` block inside of an async function and it stops execution at that line (next line is not executed) but it does not enter `catch` block either.

The test scenario is to discover the device, then turn it off so it won't accept connection but still try to connect.

It happens only when I try to connect for the first time - the next attempt catches the error fine.