dotintent/react-native-ble-plx #41: [Android]: Unable to create paired connection

joshreeves, 2016-12-13:
Hello,

I am using this library to connect a React Native app to a BLE device that requires an authenticated/paired connection. My code is very simple and works fine in iOS—I receive a system alert upon connection that the device is requesting to pair. However, the Android version does not work. The connection and service discovery steps work, but as soon as a call `writeCharacteristicWithResponseForService` I receive the following error: 

```
{ [Error: BleGattException{status=14, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_WRITE'}}]
framesToPop: 1,
code: '700',
line: 25627,
column: 21,
sourceURL: 'http://localhost:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false' }
```

Here is my code:

```
this.bleManager.connectToDevice(device.uuid)
      .then((device) => {
        console.log("DISCOVERING");
        return device.discoverAllServicesAndCharacteristics();
      })
      .then((device) => {
        console.log("DISCOVERED");
        let array = new Uint8Array([0, 0, 0, 0, 0, 0, 0, 0, 0, 0,0, 0, 0, 0, 0, 0, 0, 0, 0, 0,1, 1, 1]);
        let data = base64.fromByteArray(Buffer.from(array.buffer));
        return device.writeCharacteristicWithResponseForService('181C', '34FB', data)
        .catch((response) => {
            console.log(response);
          });
      });
```

Like I said, it's all working fine in iOS, but not Android. Any help is greatly appreciated!

Cierpliwy, 2017-11-29:
Closing this issue in favour of #187. This functionality needs to be implemented.

closed 2017-11-29 by Cierpliwy

alanyong91, 2017-12-16:
`Cannot read property 'connectToDevice' of undefined`

I get this error. How to fix this ?

Ajaykumar02, 2019-12-26:
> `Cannot read property 'connectToDevice' of undefined`
> 
> I get this error. How to fix this ?

Your bleManager instance is ,try 
import BleManager from 'react-native-ble-manager';
and call BleManager.connectToDevice

SOLR4189, 2021-01-12:
is this issue "BleGattException{status=14, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_WRITE'}}" fixed?

I get next error:

```
{
	"message": "Characteristic YYYYYYYYYYYY write failed for device XXXXXXXXXXXXX and service ZZZZZZZZZZZZZZZ",
	"errorCode": 401,
	"attErrorCode": 14,
	"iosErrorCode": null,
	"androidErrorCode": null,
	"reason": "GATT exception from MAC address XXXXXXXXXXXXX,
	status 14 (GATT_ERR_UNLIKELY),
	type BleGattOperation{description='CHARACTERISTIC_WRITE'}.
	(Look up status 0x0e here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)",
	"name": "BleError",
	"line": 24736,
	"column": 28,
	"sourceURL": "http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"
}
```

is the same issue?