dotintent/react-native-ble-plx #42: dyld: Library not loaded: @rpath/BleClientManager.framework/BleClientManager

sumesh1993, 2016-12-20:
Getting the above error. I installed as instructed but it doesn't seem to be working.
Any ideas what i might be doing wrong ?

"react": "15.4.1",
"react-native": "0.39.2",
"react-native-ble-plx": 0.3.3

sumesh1993, 2016-12-20:
@Cierpliwy

sumesh1993, 2016-12-20:
And on android i am getting this error ---- Unknown named module: 'fbjs/lib/invariant'

i tried installing fbjs but still the same error.

Please help

Muriel0603, 2017-01-09:
build path ---> link binary  -- > 添加bye目录下的三个framework

haohcraft, 2017-03-20:
@Muriel0603 你所指的是那个`run script`? 我还是会出现同样的问题。 求助，感谢。

[image]

Muriel0603, 2017-06-01:
上面两个选中 ，然后在general --> embedded binaries 添加这三个framework

Cierpliwy, 2017-06-02:
Any news on that?

closed 2017-06-07 by sumesh1993

CrazyWr, 2017-07-28:
[image]
simulator ran successfully but my iPad crashed like this

durgesh0001, 2018-02-09:
Thanks it was great help for me