dotintent/react-native-ble-plx #44: is it possible to put the device running the application in advertise mode?

vijay122, 2017-01-18:
i am creating a android application in react native. in this application i have to put the device in advertise mode, so other nearby devices (over bluetooth) can identify it and subscribe to it. i am seeing examples where we can listen to advertised devices and get its characteristics using plx, is it possible to advertise from my app?

Cierpliwy, 2017-01-18:
Unfortunately not. We still need to do a lot of work related to that on our backend libraries (RxBluetoothKit and RxAndroidBLE).

vijay122, 2017-01-23:
@Cierpliwy thankyou for the reply. can you suggest me any package which i can use for advertising in react-native. or is it a good to use android's ble advertising module in the react-native project with a wrapper.?

Cierpliwy, 2017-02-01:
Unfortunately I'm not familiar with any library that supports Peripheral Mode.

anton-aleksandrov, 2017-03-10:
why peripheral mode is a problem?

Cierpliwy, 2017-06-02:
Peripheral mode will require support for it in our RxBluetoothKit and RxAndroidBle libraries to land in react-native-ble-plx. Also Peripheral mode is working fine on Android from version 5.0 and on selected devices from what I know.

Cierpliwy, 2017-11-29:
Closing as we are blocked by missing implementation of RxBluetoothKit and RxAndroidBle libraries.

closed 2017-11-29 by Cierpliwy

robertherber, 2018-12-10:
This seems to be available in RxBluetoothKit now, right?

thePsguy, 2019-06-24:
@Cierpliwy, Should this issue be opened again now that it looks like peripheral / advertisement support is now present in the previously blocking libraries.

artiya4u, 2020-03-28:
Hi, maybe this is the time to support the peripheral / advertisement feature. I really need this feature to create an application to track people's proximity to help reduce the COVID-19 spread.