dotintent/react-native-ble-plx #47: Device.id is undefined v0.4.0

monumentyusuf, 2017-02-22:
While i scan with filtering by service uuid scanned device id property is undefined. It was working great on `v0.3.3`

Cierpliwy, 2017-02-23:
On what platform do you have an issue? I've just checked implementation on both Android and iOS and I can't find any problems on native and JS side.

monumentyusuf, 2017-02-24:
Im sorry @Cierpliwy i just forget the mention about platform. Im working with Android

Cierpliwy, 2017-02-24:
The only thing that comes to my mind right know is that your native library was not rebuild with changes which occurred between 0.3.3 and 0.4.0. Could you clean your Android project and try again? If it doesn't help, please post minimal code snippet which doesn't work correctly.

Cierpliwy, 2017-06-02:
Could you check if it works on 0.5.0?

Cierpliwy, 2017-11-29:
Closing as this is an overdue issue which should be working just fine on 0.6.5.

closed 2017-11-29 by Cierpliwy