dotintent/react-native-ble-plx #48: BleScanner example not working

readysteadygo2006, 2017-02-23:
Hi,
New to react-native, trying to get example working:
Had to make changes to versions of highlighted packages
to get example to run: 

   **"react": "15.4.1",
    "react-addons-pure-render-mixin": "15.4.1",
    "react-native": "0.38.0",**
    "redux": "3.5.2",
    "react-redux": "4.4.5",
    "redux-logger": "2.6.1",
    "react-native-router-flux": "3.32.0",
    "immutable": "3.8.1",
    "redux-immutable": "3.0.6",
    "react-immutable-proptypes": "2.0.0",
    "react-native-button": "1.6.0",
    "react-native-android-permissions": "0.0.6",
    "buffer": "4.7.1",
    "react-native-ble-plx": "Polidea/react-native-ble-plx"

originally had problem with:
error: Strict mode does not allow function declarations in a lexically nested statement on a newly created app

Now get:

"Warning: Failed prop type: The prop 'uuid' is marked as required in 'ScannedDeviceView', but its value is NULL"

Is there an updated version of example which works out of the box?

Additionally noticed article:

https://www.polidea.com/blog/ReactNative_and_Bluetooth_to_An_Other_level/

Is code for this exampel available on github?

Are there any other examples on github or similar which would give me some 
working exampels to look at?

Thanks for any help you can provide!

BR
Paul

Cierpliwy, 2017-02-23:
Hello,

Library is still not marked as stable and we want to fine tune API before stabilisation. In version 0.4.0 all device.uuid, service.deviceUUID, characteristic.deviceUUID properties names were changed to: device.id, service.deviceID, characteristic.deviceID. Did you make changes? 

PS: I suggest to use library from npm and not directly from git repository.

readysteadygo2006, 2017-02-24:
Cheers, that was the problem. All working now!
Thanks

closed 2017-02-24 by readysteadygo2006