dotintent/react-native-ble-plx #52: Not able to bundle js

srinuch1981, 2017-03-15:
MINGW64 ~/Downloads/ReactBLEScanner $ react-native bundle --platform android --dev false --entry-file index.android.js   --bundle-output android/app/src/main/assets/index.android.bundle   --assets-dest android/app/src/main/res/ Unable to parse cache file. Will clear and continue. [3:18:37 PM] <START> Building Dependency Graph [3:18:37 PM] <START> Crawling File System [3:18:37 PM] <START> find dependencies [3:18:48 PM] <END>   Crawling File System (10967ms) [3:18:48 PM] <START> Building in-memory fs for JavaScript [3:18:48 PM] <END>   Building in-memory fs for JavaScript (251ms) [3:18:48 PM] <START> Building in-memory fs for Assets [3:18:48 PM] <END>   Building in-memory fs for Assets (188ms) [3:18:48 PM] <START> Building Haste Map [3:18:49 PM] <START> Building (deprecated) Asset Map [3:18:49 PM] <END>   Building (deprecated) Asset Map (89ms) [3:18:49 PM] <END>   Building Haste Map (385ms) [3:18:49 PM] <END>   Building Dependency Graph (11798ms) [3:19:06 PM] <END>   find dependencies (28496ms) bundle: start bundle: finish bundle: Writing bundle output to: android/app/src/main/assets/index.android.bundle C:\Users\schava\Downloads\ReactBLEScanner\node_modules\promise\lib\done.js:10       throw err;       ^

Cierpliwy, 2017-06-02:
Any news on that?

Cierpliwy, 2018-04-09:
Closing due to inactivity

closed 2018-04-09 by Cierpliwy