dotintent/react-native-ble-plx #54: Runtime Error

srinuch1981, 2017-03-15:
[image]

vldgeorgiev, 2017-03-21:
I got the same one on Android. With the latest Node and ReactNative as of now. Also tried updating the "react-native" module, but that of course lead to other errors.

haohcraft, 2017-03-22:
I think sharing some code in this case would be helpful for others to look into.

vldgeorgiev, 2017-03-22:
Actually I was referring to the example app from /examples/ReactBLEScanner/ in the repo. The error appears when the app is built and started. Probably it has worked with a previous version of ReactNative, but not with the current one 0.42.
I just missed the fact that the original poster has not specified how he got the error.

vldgeorgiev, 2017-03-22:
Here is a solution, which involves modifying some of the react-native sources https://github.com/facebook/react-native/issues/11389#issuecomment-287897612
In the same thread there are fixes for newer react-native versions too.

Cierpliwy, 2018-04-09:
Old example is not supported. Closing due to inactivity.

closed 2018-04-09 by Cierpliwy