dotintent/react-native-ble-plx #55: Command /bin/sh failed with exit code 1

WillDent, 2017-03-17:
Unrecognized arguments: "/Users/willdent/Library/Developer/Xcode/DerivedData/ble-ausclppugcjrecboiwdlaycdoyfq/Build/Intermediates/ble.build/Debug-iphoneos/ble.build/Script-5D4CFAB11E7BFAE5005783F4.sh"
Command /bin/sh failed with exit code 1

WillDent, 2017-03-21:
@haohcraft maybe we can help one another?  Are you pulling from master or using what is in NPM?

I tried with npm.  Guess I will try from master.

haohcraft, 2017-03-22:
@WillDent so I have resolved this. I had put the `copy-frameworks` script in the wrong place. Could you share me with a screenshot of your Run Script setting in xcode? I could take a look.

tuantle, 2017-03-23:
Hi, I have the same issue. And this what I have set.
[image]

haohcraft, 2017-03-24:
1. make sure `carthage` has installed
2. make sure the frameworks have been in `node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS/`, otherwise you should re npm install the lib
3. and my `run script` looks like below:

[image]

tuantle, 2017-03-24:
Thanks. That fixed it for me, I originally put the copy-frameworks line in the Shell input box.

WillDent, 2017-03-25:
@haohcraft @tuantle got this working. thanks

closed 2017-03-25 by WillDent

ykcai, 2017-04-25:
I am getting ```[34;1m***[0m Please update to the latest Carthage version: 0.21.0. You currently are on 0.20.1
Failed to read file or folder at /Users/michaelcai/Desktop/React/IoTApp/ios/../node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS/RxBluetoothKit.framework

Command /bin/sh failed with exit code 1```

Cierpliwy, 2017-04-26:
Probably you didn't build iOS frameworks properly. Does carthage finish successfully when react-native-ble-plx is installed? Please check all manual steps.

nickvido, 2017-08-14:
I'm having the same problem. The specified folder exists:

```
$ ls -la node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS/RxBluetoothKit.framework
total 12672
drwxr-xr-x   6 myname  myrole      204 Aug 13 20:46 .
drwxr-xr-x  24 myname  myrole      816 Aug 13 20:46 ..
drwxr-xr-x   3 myname  myrole      102 Aug 13 20:46 Headers
-rw-r--r--   1 myname  myrole      769 Aug 13 20:46 Info.plist
drwxr-xr-x   4 myname  myrole      136 Aug 13 20:46 Modules
-rwxr-xr-x   1 myname  myrole  6480052 Aug 13 20:46 RxBluetoothKit
```

Version information:
```
"react": "16.0.0-alpha.12",
"react-native": "^0.46.1",
"react-native-ble-plx": "^0.6.1",
XCode: 8.3.3 (8E3004b)
```

nisargap, 2017-10-30:
Same issue

nisargap, 2017-10-30:
@nickvido did you ever figure it out?

nisargap, 2017-10-31:
@nickvido  I fixed that problem by using an absolute path for RxBluetoothKit.framework

rynop, 2017-10-31:
I get the same error as well, but on `Carthage/Build/iOS/RxSwift.framework`.  If I hard code input files in `Run Script` like @nisargap does I do not get the error.  Folder exists.

```
xcode Version 9.1 (9B55)
react@16.0.0-alpha.6
react-native@0.44.0
react-native-ble-plx@0.6.4
carthage version 0.26.2
```

FWIW I also ran into issue on Xcode v9.0.  I've tried cleaning project (cmd+opt+shift+k) no use.

@Cierpliwy any idea what is going wrong? Hard code kills team development.

nickvido, 2017-11-01:
I ended up scrapping the whole thing and re-creating the app using `react-native init` instead of `create-react-native-app`. I also switched to just using the `react-native-ble-manager` package that this package references.

rynop, 2017-11-01:
@nickvido thanks. Curious, where do you see `react-native-ble-manager` referenced to in this project? I dont see it in `package.json` and don't find any codehits. Just wondering what the diff is between this project and `react-native-ble-manager`.

The other piece of info I did not mention yesterday, was my project was started with this boilerplate: https://github.com/junedomingo/movieapp so wonder if not starting clean as you mention has something to do with this issue.

vadimue, 2017-12-12:
@haohcraft Thanks, you save my time!

seriph, 2018-01-13:
@rynop @nickvido The issue here actually turned out to be caused by copying these Input Files from the install instructions. It copied a newline character due to the way the md file is formatted. Close inspection of the logs (reading through all the environment variables) revealed this.
@Cierpliwy I think we should place a note warning against this in the install instructions or reformat it to be less likely.

Pouria22, 2018-07-14:
I had this issue because i denied keychain access