dotintent/react-native-ble-plx #56: new BleManager() failed on componentWillMount

haohcraft, 2017-03-21:
In my own component below, I try to play with the ble but get some error:

```
undefined is not an object (evaluating 'BleModule.createClient')
```

Looks like in BleManager.js, it can not import BleModule. Any idea?

```
import { BleManager } from 'react-native-ble-plx';

export default class Discover extends Component {

	constructor(props) {
	  super(props);
	
	  this.state = {
	  	device: { id: '', name: '' },
	  	scanning: false
	  };
	}
	componentWillMount() {
	    this.manager = new BleManager();
	    this.subscriptions = {}
	    this.manager.onStateChange((newState) => {
	      console.log("State changed: " + newState)
	    });
	}
	componentDidUpdate(prevProps, prevState) {
		if (this.state.scanning) {
			this.manager.startDeviceScan(null, null, (error, device) => {
				if (!error) {
					console.log("Found device: ", device);
					this.setState({
						device,
						scanning: false
					})
				}
			});
		}
	  
	}
	componentWillUnmount() {
	    this.manager.destroy();
	    delete this.manager;
  	}
	render() {
		const { device } = this.state;
		return (
			<View style={{ flex: 1, justifyContent: 'center' }}>
				<Text>
				  {device.id} -- {device.name}
				</Text>
				<ButtonView
					onClick={this._startScan.bind(this)}
					text={'Start scanning'}
					color={'#beffc6'}/>
          
			</View>
		);
	}
	_startScan() {
		this.setState({ scanning: true });
	}
};

```

haohcraft, 2017-03-22:
I have resolved this. It was due to a bad configuration on my side.

closed 2017-03-22 by haohcraft

JordanKlaers, 2017-10-21:
what was the correct configuration? I am having the same issue and am having trouble understanding it.

Benjamin-Lin-X, 2017-11-11:
I am having the same issue while using expo too. Please tell me how to fix it.

Benjamin-Lin-X, 2017-11-11:
@haohcraft I am having the same issue while using expo too. Please tell me how to fix it.

brancooo1, 2018-02-07:
Guys, to solve it, just go through the whole manual installation of ble-plx
https://github.com/Polidea/react-native-ble-plx#manually
Make sure, from point 1, that everything is as it should be by documentation.

It'll work ;)