dotintent/react-native-ble-plx #57: bash

ecasona, 2017-03-22:
npm ERR! react-native-ble-plx@0.4.0 postinstall: `bash build_ios_frameworks.sh`
npm ERR! Exit status 1

ecasona, 2017-03-22:
what can i do

haohcraft, 2017-03-22:
Have you installed  Carthage before running npm install ?

ecasona, 2017-03-23:
@haohcraft  I have tried npm install carthage  ,but the problem exits already.
 > react-native-ble-plx@0.4.0 postinstall E:\react native\work_place\test_ble\node_modules\react-native-ble-plx
> bash build_ios_frameworks.sh

'bash' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

I am in windows dev envirenment。

m7775223, 2017-04-11:
I also encountered this problem in windows dev envirenment。 What else do you do before you install the npm install react-native-ble-plx？

Cierpliwy, 2017-06-02:
On Windows you have to have bash in system path to install this library. It is something which could be improved.

Cierpliwy, 2018-04-09:
Bash is not used anymore in build process.

closed 2018-04-09 by Cierpliwy