dotintent/react-native-ble-plx #60: Characteristic monitoring causes delayed execution of fetch and other Promises

vldgeorgiev, 2017-04-17:
I am using the library to monitor a BLE device for data, and after some amount is accumulated I upload it to a REST service.
It mostly works well in the device discovery, connection and writing data. However, there is an issue with the response received from the monitored characteristic.
I am not sure how to describe it well, but essentially any `fetch` calls or other Promises don't get resolved properly. The `.then()` or `.catch()` clauses after the fetch are called with a delay only after the BLE device disconnects. In most cases the fetch finishes with a 'Network request failed' error.

Here is some sample code to make it more clear:
I set up a monitor with:
```
    const subscription = bleManager.monitorCharacteristicForDevice(deviceId,
      'some_specification', 'some_characteristic', handleResponse, null);
```
The `handleResponse` function dispatches an async action using `redux-thunk`, which calls fetch to upload the REST data.
```
function handleResponse(error, characteristic) {
  // The action below contains a simple fetch().then() call. It is dispatched with 'redux-thunk'. However
  // in most cases the .then() executed with a very large delay, or the whole fetch ends with an error,
  // again after a delay 
  store.dispatch(uploadAction(somedata));

  // A direct call to fetch here behaves like the action above
  fetch('https://www....')
    .then((response) => { ... })
    .catch((err) => { ... });
}
```
Seems that any promises called directly or through actions from the handleResponse get resolved only after the BLE device is disconnected. If I disconnect it explicitly after sending the action, it all works well.
The same actions or fetch calls elsewhere in the app work properly, so I suppose this is some interaction with the BLE library that is causing the issue.

I would be grateful for any ideas how to solve that.

bcohn12, 2017-04-17:
I'm trying to build a similar service for monitoring a Bluetooth continuous signal. Hope you figure this out!

vldgeorgiev, 2017-04-18:
For the moment I have put it on hold, after trying several approaches with events or actions.
Other than that, the library is working very well, so it is possible that the problem is due to some wrong usage on my part.

closed 2019-03-23 by stale[bot]