dotintent/react-native-ble-plx #61: Support duplicate services with same UUID

brycejacobs, 2017-04-17:
Please see Polidea/RxBluetoothKit#82

Currently, there is no way to write to a service / characteristic that shares a UUID. This is somewhat of a showstopper for me. Is there anything I can do to get this moving along? 

I know this is dependent on the RxBluetoothKit, but it's worth considering the changes that will need to take place here in order to gain support.

brycejacobs, 2017-05-01:
Anybody home?

Kacper20, 2017-05-04:
Hello @brycejacobs ,
It's already merged in RxBluetoothKit to 3.0.8 version of the library.
We will integrate this changes straight into `react-native-ble-plx` library - this will happen probably on Monday, since this week is full of bank holidays in Poland and many of us are on vacations.

Thanks for your patience,
Kacper

brycejacobs, 2017-05-04:
@Kacper20 Thanks! Looking forward to testing it out!

Cierpliwy, 2017-05-10:
We need one more change in RxBluetoothKit library to implement it in react-native-ble-plx. I allocated some time to do just that in this and following week.

brycejacobs, 2017-05-17:
@Cierpliwy I see that you merged in Polidea/RxBluetoothKit#111

Was this the fix you were referring to?

brycejacobs, 2017-05-17:
Btw, have you thought of how this is going to affect calls that rely on the string UUID?

For example:  manager.characteristicsForDevice(deviceIdentifier: string, serviceUUID: string)

Would it be better served to change serviceUUID->objectId that you merged in to the RXBluetoothKit?

Cierpliwy, 2017-05-17:
Hello,

Merge which you mentioned should be enough to support services and characteristics with the same UUUID. The plan is to add new id field to Service and Characteristic classes.

There will be new _additional_  API to read/write/notify directly without a need to remember UUID fields like:
```js
device.readCharacteristicForDevice(deviceIdentifier: string, characteristicID: number, transactionId: string): Promise<Characteristic>
service.readCharacteristic(characteristicID: number, transactionId: string): Promise<Characteristic>
```

There are two options:
1. Overload functions by type checking arguments (number for id, string for uuid)
2. Create new functions with suffix like: *ById.

Maybe you have some other suggestions as well? Which option do you like the most personally?

brycejacobs, 2017-05-17:
Option 1, but I think leaving the option to pass a string UUID to the API is a mistake, and will confuse people who aren't familiar with the reason why the Id field is being added in the first place.

For example, your surface API will act differently whenever there are multiple UUID's that are identical but the user still invokes the old UUID api call.

```service.readCharacteristic(characteristicUUID: string, transactionId: string)```
Does the service read the first characteristic with a matching UUID? I can see this causing confusion.

I think switching over to the ID field makes more sense for the long run, but it's your call. Either way will work, just don't see any reason why invoking with the UUID over a generated ID is more attractive.

Another option is to instantiate a UUID class object. However, passing an instantiated object to the surface API in a scripting language seems clunky.

Cierpliwy, 2017-05-17:
There are two nice things which we will have by keeping API which supports UUIDs:

1). We don't need to search for our characteristics by calling `services()` and `characteristics()` functions. Generated Ids will not be stable between application runs so we need to do this step every time and in most cases characteristics will be cached on user side where the same thing is already done on native side.

2) I have never worked with BLE devices which used characteristics with the same UUIDs. So by keeping UUID APIs we are just making it simpler for the most common case.

brycejacobs, 2017-05-17:
Then I would say just stick with type checking with a disclaimer in the Docs. 

Our use case revolves around a hub, that has multiple devices that are represented by the same service descriptions. Which has become somewhat popular with the mesh spec taking so long to be released.

Cierpliwy, 2017-05-24:
I have done most of the work (branch ver_0_5) to support services and characteristics with the same UUID. Currently I updated iOS native module and tomorrow will clean up code and test it more thoroughly. Then I will implement required changes in Android's native module.

By this time if you have time, please check if implementation is working fine for you. There were some changes and API decisions along the process:

* API remained stable - I didn't introduce any new calls which are public.
* There are _private_ functions which will accept service/device IDs and they are used when you are calling convenience functions (like below). Performance should be better as well.

So to take advantage of this work you should:
- call device.services() (remember results)
- call service.characteristics() (remember results)
- call characteristic.read(), characteristic.writeWithResponse() etc.

Above functions will use IDs only. There is also new `id` field in Service and Characteristic classes for your disposal - you can use them to uniquely identify BLE objects.

brycejacobs, 2017-05-24:
Perfect timing, let me run this through the ringer a little and see what happens. I may not have time to do it today, but no later than Friday.

Cierpliwy, 2017-06-02:
Changes landed in branch 0.5.0. If something is not working properly please open another issue.

closed 2017-06-02 by Cierpliwy