dotintent/react-native-ble-plx #62: 'Incompatible pointer types' error in initWithQueue

jl-mindius, 2017-04-19:
Hi!

We have successfully used this library in a project for a while. After a month of hiatus we got back to it this week and can't run it on iOS anymore. The only change I can think of is an update to Xcode 8.3.1.

It builds successfully, but when creating a new client with `new BleManager()` it crashes with a `SIGABRT`. Xcode has a warning:

`node_modules/react-native-ble-plx/ios/BleClient.m:39:56: Incompatible pointer types sending 'dispatch_queue_t' (aka 'NSObject<OS_dispatch_queue> *') to parameter of type 'OS_dispatch_queue * _Nonnull'`

Pointing here:

```
RCT_EXPORT_METHOD(createClient) {
    _manager = [[BleClientManager alloc] initWithQueue:self.methodQueue];
    _manager.delegate = self;
                        ^
}
```

Which leads to another error:

`node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS/BleClientManager.framework/Headers/BleClientManager-Swift.h:131:69: Passing argument to parameter 'queue' here`

Pointing here:

```
SWIFT_CLASS("_TtC16BleClientManager16BleClientManager")
@interface BleClientManager : NSObject
@property (nonatomic, strong) id <BleClientManagerDelegate> _Nullable delegate;
- (nonnull instancetype)initWithQueue:(OS_dispatch_queue * _Nonnull)queue OBJC_DESIGNATED_INITIALIZER;
                                                                    ^
```

Any ideas?

Cierpliwy, 2017-04-19:
Hello there. We are successfully build and use this library in our internal project without any issues on XCode 8.3.1. Make sure that you rebuild all used frameworks by react-native-ble-plx, by reinstalling it. If it is a runtime error I assume that there is some binary incompatibility between libraries.

jl-mindius, 2017-04-19:
I feel pretty stupid for not trying that, it works now, thanks!

For anyone else: just do `npm uninstall react-native-ble-plx`, `npm install react-native-ble-plx`.

closed 2017-04-19 by jl-mindius