dotintent/react-native-ble-plx #64: Runtime Error: SIGABRT on fileprivate var connectedDevices = Dictionary<UUID, Peripheral>()

watadarkstar, 2017-05-03:
I get this runtime error `Runtime Error: SIGABRT` on `fileprivate var connectedDevices = Dictionary<UUID, Peripheral>()` in `BleClientManager`:

[image]

My code is:

```
class BluetoothList extends Component {
    componentWillMount() {
        this.manager = new BleManager(); // <------ HERE
        this.manager.onStateChange((newState) => {
          console.log("State changed: " + newState)
        })
        console.log('will mount test');
    }

    componentWillUnmount() {
        this.manager.destroy();
        delete this.manager;
    }

    render() {
       ...
    }
}
```

I am running on my device using [Expo](https://expo.io/) as a detached app. My dependencies are:

```
"dependencies": {
    "expo": "16.0.0",
    "react": "16.0.0-alpha.6",
    "react-native": "https://github.com/expo/react-native/archive/sdk-16.0.0.tar.gz",
    "react-native-ble-plx": "^0.4.0",
    "react-native-elements": "^0.11.2"
  },
```

seanadkinson, 2017-05-05:
FYI, I was having the same issues, and fixed it by rebuilding all carthage artifacts.  I think this issue occurs if you have upgraded xcode, and still had artifacts built from a previous version.

watadarkstar, 2017-05-06:
@seanadkinson I actually did upgrade xcode recently while dealing with these issues. What do you mean by rebuilding all carthage artifacts? How do I do that exactly?

seanadkinson, 2017-05-08:
For me:

    $ cd ios
    $ rm -Rf Carthage
    $ carthage bootstrap --platform iOS

Cierpliwy, 2017-05-10:
Alternatively from root of your project you can execute below commands

```bash
cd node_modules/react-native-ble-plx/
./build_ios_frameworks.sh
```

to force rebuild of native libraries. Let me know if it helped.

watadarkstar, 2017-05-11:
@Cierpliwy Yup that worked! Thank you.

closed 2017-05-11 by Cierpliwy