dotintent/react-native-ble-plx #65: Response for writeWithResponse

dhruvp, 2017-05-04:
Hi!

First of all, thank you for the wonderful library! It's been a pleasure to use so far.

I'm currently trying to write to a given characteristic with response but the library seems to just return the source characteristic and not provide the success or error response returned by the peripheral.

For instance, Apple's core bluetooth library returns if there is an error during the write [in the didWriteValueForCharacteristic function]( https://developer.apple.com/reference/corebluetooth/cbperipheraldelegate/1518823-peripheral?language=objc).

Is there a way we can get something similar in this library? I tried digging through the source code but got lost at $0.writeValue as I wasn't able to see where that was defined.

Thank you! This would help a ton!

Cierpliwy, 2017-05-10:
On iOS side you have following code to handle write with response:

```swift
let disposable = characteristicObservable(deviceId,
                                                  serviceId: serviceId,
                                                  characteristicId: characteristicId)
            .flatMap {
                $0.writeValue(value, type: response ? .withResponse : .withoutResponse)
            }
            .subscribe(
                onNext: { characteristic in
                    promise.resolve(characteristic.asJSObject)
                },
                onError: { error in
                    error.bleError.callReject(promise)
                },
                onCompleted: nil,
                onDisposed: { [weak self] in
                    self?.transactions.removeDisposable(transactionId)
                    BleError.cancelled().callReject(promise)
                }
            )
```

This code basically is finding your characteristic based on UUIDs, writes value and then resolves when successful or emits error.

In RxBluetoothKit writeValue is using monitorWrite function:

```swift
 public func writeValue(_ data: Data,
                           for characteristic: Characteristic,
                           type: CBCharacteristicWriteType) -> Observable<Characteristic> {
            return .create { observer in
                let disposable: Disposable
                switch type {
                case .withoutResponse:
                    disposable = self.ensureValidPeripheralState(for: .just(characteristic)).subscribe(observer)
                case .withResponse:
                    disposable = self.ensureValidPeripheralState(for: self.monitorWrite(for: characteristic).take(1))
                        .subscribe(observer)
                }
                self.peripheral.writeValue(data, for: characteristic.characteristic, type: type)
                return disposable
            }
    }
```

which wraps callback which you mentioned:

```swift
public func monitorWrite(for characteristic: Characteristic) -> Observable<Characteristic> {
        return peripheral
            .rx_didWriteValueForCharacteristic
            .filter { return $0.0 == characteristic.characteristic }
            .map { (_, error) -> Characteristic in
                if let error = error {
                    throw BluetoothError.characteristicWriteFailed(characteristic, error)
                }
                return characteristic
            }
    }
```

So from my perspective code handling characteristic writes looks fine. What's more, it is working fine with our BLE devices. Remember that characteristic in JS form don't update written value, so it may look like write didn't succeed!

alexandrapersea, 2017-05-22:
I am facing a similar problem. When i try writing to a characteristic the returned characteristic has no value.

This is my call:

`this.manager.writeCharacteristicWithResponseForDevice('device_id',
                                                                'service_id',
                                                                'characteristic_id',
                                                               'base64_command')
     .then(value => console.log(value))
       .catch(err => console.log(err))`

The returned characteristic:

`Characteristic {
  deviceID: "device_id"
  isIndictable: true
  isNotifiable: false
  isNotifying: false
  isReadable: false
  isWritableWithResponse: true
  isWritableWithoutResponse: false
  serviceUUID : "service_uuid"
  uuid:  "characteristic_uuid"
  value: null
  _manager: BleManager
  __proto_ : NativeCharacteristic }`

Cierpliwy, 2017-05-22:
By default CBPeripheral on iOS doesn't update Characteristic internal value after write requests but only when read is invoked. This is totally different behaviour then Android. I guess you find it strange and I plan to return value in such cases in coming version 0.5.0.

By this time I recommend to just assume that passed value to  `writeCharacteristicWithResponseForDevice` was written.

alexandrapersea, 2017-05-22:
Ok, is there a way to get the response of the device atm? I need to pass the success/error message to the user.

Cierpliwy, 2017-05-22:
What do you mean by response of the device? When `writeCharacteristicWithResponseForDevice` succeeded (as a Promise.resolve) you are certain that value was passed to your connected device's characteristic. You can always call successive `readCharacteristicForDevice` to check that if you want and characteristic supports it (`characteristic.isReadable`).

alexandrapersea, 2017-05-22:
Okay, got it. 

Thank you for your help and great job with this library! 👍

Cierpliwy, 2017-05-22:
No problem! Closing this issue for now, please reopen if needed.

closed 2017-05-22 by Cierpliwy