dotintent/react-native-ble-plx #67: iOS Library not loaded: @rpath BleClientManager

kinber95, 2017-05-04:
I get following error message if I run on my iOS device:

`dyld: Library not loaded: @rpath/BleClientManager.framework/BleClientManager
  Referenced from: /var/containers/Bundle/Application/2674B4D0-D879-4940-84B3-3CE7994CFA82/ishareit.app/ishareit
  Reason: image not found`

My app freeze at the start screen. On android I had no problems, any ideas ?

closed 2017-05-04 by kinber95

jcady, 2017-06-13:
The problem for me was that in `Build Phases > Run Script` I had the checkbox `Run script only when installing` checked for the carthage copy-frameworks script. That needs to stay unchecked.

rknell, 2017-12-01:
I needed to do a project clean and rebuild, HTH

swaroopa94, 2018-03-09:
@jcady  
Hi I did the sma ething but the error still existe in xcode 9.1
But working fine in xcode 8.3.2
Can you suggest how to make it work in xcode 9.1