dotintent/react-native-ble-plx #68: Unknown error

adyanced, 2017-05-13:
Hi, thanks for this wonderful library.
I encountered a error when I disconnect from ble device (Sorry I failed upload the snapshot):

Unknown error:
com.polidea.rxandroidble.exceptions.BleDisconnectedException

But everything is ok while running your example apk from google play store.
More info:
React native 0.44.0 (upgraded from 0.43 after the same issue )
Android buildToolsVersion 25.0.2
minSdkVersion 18
targetSdkVersion 23

adyanced, 2017-05-13:
I found the reason. If back from CharacteristicDetailsView but not off the notification, the unknown error is throwed

Cierpliwy, 2017-05-14:
In such cases I recommend to use nRF Connect (Android) or LightBlue (iOS) to verify if these apps are also emitting same or similar errors.

adyanced, 2017-05-14:
Thanks for reply.
I modified Sniffator demo code. when a click on scanned device it will directly jump to CharacteristicDetailsView which skipped off ServiceView and CharacteristicView. And when a BACK button push, it returns to scannedDeviceView. Thus an Unknown error is throwed.

Cierpliwy, 2017-05-14:
I would not recommend looking at or using Sniffator example app as it is outdated and using mistakenly newest version of react-native-ble-plx. I need to update it. 

The only requirement from users perspective is to call `connectToDevice ` and `discoverAllServicesAndCharacteristicsForDevice`.  Also you don't need to cancel or turn off anything before explicit or implicit disconnection from device.

Did I answer your question?

Cierpliwy, 2017-06-02:
Any news on that?

adyanced, 2017-07-29:
Hi, me again.
Sniffator is just an evaluation. 
I write my code after I study RN cause it's new to me. But the error stay there, if disconnect from ble while notification is enabled, so I

closed 2019-03-23 by stale[bot]