dotintent/react-native-ble-plx #70: Classic bluetooth detection

marwakhalil, 2017-05-15:
Hello, 
I'm new in react native, i used your library which is amazing by the way, but the problem that i couldn't detect devices that doesn't support BLE,
is it possible to detect it or not ?

Cierpliwy, 2017-05-15:
This library supports and will support only BLE devices.

marwakhalil, 2017-05-17:
Okay thanks, 
but can i connect to a smartphone advertised as iBeacon, because i keep on having this message on the error zone: BleGattException{status=133 ,bleGattOperation ...  ??? 

I'm using the Sniffator example,

marwakhalil, 2017-05-18:
[image]

stinodes, 2017-05-18:
@marwakhalil I am connecting to an Android device which is acting as a peripheral, and am also getting status=133
I do not get that error, or at least not 100% of the time, using just native code. I don't really know what the difference is that causes this.

marwakhalil, 2017-05-18:
The thing that i found out that advertising my android as a peripheral, it generates the mac id "i use quickBeacon app to advertise"  and i have never been able to connect to it, i don't really know what to do

stinodes, 2017-05-18:
@marwakhalil I have been browsing through native code, and try to align a sample app I have here with what is used in the RxBle library, and I found that, in my case, `device.connectGatt(this, false, mGattCallback, BluetoothDevice.TRANSPORT_LE);`, the last parameter was the culprit.  
If I added the `TRANSPORT_LE` to my sample Android app, it 100% returned the same error, whereas otherwise it would work nearly all the time.

In my previous project, on the other hand, this did not seem to be a problem, so it might be related to the fact that you (or we) are connecting to a smartphone, rather than, say, a Nordic board.

I am not sure whether this can be fixed though. It seems more like an edge case, but it would be great if this was configurable (for me at least, not generally per se 😄 ).

marwakhalil, 2017-05-18:
Okay, is there any chance that i can send data a json file to a DA14680 chip ?

Cierpliwy, 2017-06-02:
Closing this specific issue about classic Bluetooth support. If you have another problem please open new issue.

closed 2017-06-02 by Cierpliwy