dotintent/react-native-ble-plx #71: Error redux-logger v3

Carole89, 2017-05-17:
Please i need help! 

when i run ReactBLEScanner application in my android emulator I have this problem and i did not understand how I can correct this error

[image]

package.json 

```
{
  "name": "ProjBle",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "buffer": "^5.0.6",
    "immutable": "^3.8.1",
    "react": "16.0.0-alpha.6",
    "react-addons-pure-render-mixin": "^15.5.2",
    "react-immutable-proptypes": "^2.1.0",
    "react-native": "0.44.0",
    "react-native-android-permissions": "0.0.7",
    "react-native-ble-plx": "^0.4.0",
    "react-native-button": "^1.8.2",
    "react-native-router-flux": "^3.39.1",
    "react-redux": "^5.0.4",
    "redux": "^3.6.0",
    "redux-immutable": "^4.0.0",
    "redux-logger": "^3.0.1"
  },
  "devDependencies": {
    "babel-jest": "20.0.1",
    "babel-preset-react-native": "1.9.2",
    "jest": "20.0.1",
    "react-test-renderer": "16.0.0-alpha.6",
    "remote-redux-devtools": "^0.5.0"
  },
  "jest": {
    "preset": "react-native"
  }
}

```

Cierpliwy, 2017-05-18:
Example application needs to be updated as it was developed around 0.2 version of react-native-ble-plx. Got it on my TODO list. I would recommend to start your own project from scratch right now.

Carole89, 2017-05-18:
@Cierpliwy thank you for your reply

I corrected this error by installing the version (2.6.1) but after changing the version I have another problem! :(

[image]

Cierpliwy, 2018-04-09:
Old example is not supported anymore. Closing.

closed 2018-04-09 by Cierpliwy