dotintent/react-native-ble-plx #72: Sending Data

marwakhalil, 2017-05-18:
Can i send data through this libraries,
I am connecting to a DA14680 chip through react native ble plx app and i want to send a json file 
is that possible ?
Thanks

Cierpliwy, 2017-05-18:
I have never worked with DA14680 but it looks like it supports BLE, so it's doable. Just remember that characteristic reads and writes are limited to negotiated MTU which is lower then 512B. I would suggest to use JSON binary format to compress data as much as possible (like MessagePack) to send as little chunks as possible via characteristics.

marwakhalil, 2017-05-18:
Okay this is great, sufficient for my needs  
and i would be glad if you give me the name of the method to send data
Thanks

marwakhalil, 2017-05-18:
@Cierpliwy  could you please give me the name of the method that allow sending data in this library? 
i really need it :(

Cierpliwy, 2017-05-18:
I recommend to read intro page of documentation and check functions in it:
https://polidea.github.io/react-native-ble-plx/

briancohn, 2017-05-25:
@marwakhalil did you get it to work?

marwakhalil, 2017-05-29:
@briancohn , yes i used writeCharacteristicWithResponseForDevice() and it worked

closed 2017-06-02 by Cierpliwy

0lide, 2017-06-06:
@marwakhalil, can you show example how you did it?

briancohn, 2017-06-06:
@marwakhalil I'd really appreciate that too, thanks so much

marwakhalil, 2017-06-07:
I made a scan for all devices and i fixed a mac id for test writing
`scan() {
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            console.log("something went wrong");
            return;
        }
        console.log("device id est : " + device.id + "device name  : " + device.name + "device is connectable : " +
            device.isConnectable + "rssi : " + device.rssi);
        if (device.id === ‘XX:XX:XX:XX:XX:XX’) {
		 this.manager.stopDeviceScan();
	device.connect()
   		 .then((device) => {
        	return device.discoverAllServicesAndCharacteristics();
    })

        }

    })
}

send() {
this.manager.writeCharacteristicWithResponseForDevice(deviceId,
                                                      serviceId,
                                                      characteristicId,
                                                      base64Value)
        .catch((error) => {
            console.log('error in writing data');
            console.log(error);

        })
}

`

marwakhalil, 2017-06-07:
`scan() {
this.manager.startDeviceScan(null, null, (error, device) => {
if (error) {
console.log("something went wrong");
return;
}
console.log("device id est : " + device.id + "device name : " + device.name + "device is connectable : " +
device.isConnectable + "rssi : " + device.rssi);
if (device.id === ‘XX:XX:XX:XX:XX:XX’) {
this.manager.stopDeviceScan();
device.connect()
.then((device) => {
return device.discoverAllServicesAndCharacteristics();
console.log('connected');
})

    }

})
}

send() {
this.manager.writeCharacteristicWithResponseForDevice(deviceId,
serviceId,
characteristicId,
base64Value)
.catch((error) => {
console.log('error in writing data');
console.log(error);

    })
}

`

i made a scan for all devices and i fixed a mac id for testing

shonie2xx, 2023-01-08:
@Hurobaki I know it's 2023, but have you found a solution to your question? I have the same trouble now.

lchenfox, 2023-05-12:
So how to get the serviceId  and characteristicId? If someone knows about this, please let me know. Thank U so much!