dotintent/react-native-ble-plx #75: Can't disconnect from device (Android)

kinber95, 2017-05-22:
Hello 
I want to disconnected from my Android device. I used following code:
First 
`  cancelDevice(id) {
    this.manager.cancelDeviceConnection(id)
      .then((device) => {
        console.log('Connection lost: ' + device.isConnectable + " with " + device.name);
      }, (error) => {
        console.log('cancel: ' + error);
      }) 
  }`

Second
`  cancelDevice() {
      this.device.cancelConnection()
        .then((device) => {
          console.log('Connections cancel: ' + device.name + ' ' + device.id);
        })
  }`

The manager disconnect and I cant use other functions of my code because, I am disconnected from my device according to the manager, but I can see on my Android device thats still a connection.

Any solutions ?

Cierpliwy, 2017-06-02:
Please check on version 0.5.0 as it has updated Android dependency.

Cierpliwy, 2018-04-09:
Please retest with latest version of library and reopen if needed.

closed 2018-04-09 by Cierpliwy

miawils, 2018-06-25:
Hi I am currently having this issue with the Google Pixel 2.