dotintent/react-native-ble-plx #76: Possible Unhandled Promise Rejection

marwakhalil, 2017-05-22:
Hello, I'm trying to write on my mac advertised as beacon and i'm not able to i keep on getting this warning meanwhile i'm connected to the advertised mac, 
[image]

My code looks something like this : 
 

` scanAndConnect() {
        this.manager.startDeviceScan(null, null, (error, device) => {
            if (error) {
                console.log("error");
                return;
            }

            console.log("device id est : "+device.id);
            console.log("device name est : "+device.name);
            console.log("base 64 encoding : " + encoded);
     if (

                device.name === 'MacBook Pro'    ||
                device.id === '6C:40:08:AE:7A:19'

            ) {
                console.log("this is it");
                this.manager.stopDeviceScan();

                // Proceed with connection.
                device.connect()
                    .then((device) => {
                        return "those are all characteristics"+ device.discoverAllServicesAndCharacteristics();

                    })
                    .then((device) => {
                        console.log(device.manager);
                    })
                    .catch((error) => {
                        // Handle errors
                    });

                device.writeCharacteristicWithResponseForService('6b401f54-6fca-4dbf-bb9c-b15bb7ab0958',null ,encoded,'1' );

            }
        });

    }`

Cierpliwy, 2017-05-22:
I have never worked with iBeacons. If you give me hints how to setup environment for this then I will try to help. Please also check that your device.isConnectable - if value is missing then we are not sure.

marwakhalil, 2017-05-22:
i'm currently advertising my mac using BeaconOsX app : that i got it from here https://github.com/falkorichter/BeaconOSX/releases
and for device.isConnectable the value is null but when i downloaded the Sniffator example from play store i got a connection between my android phone and my mac as beacon, 
hope you can help

marwakhalil, 2017-05-22:
@Cierpliwy  any ideas ?

closed 2019-03-23 by stale[bot]