dotintent/react-native-ble-plx #77: Can you indicate GattCharacteristics

alexandrapersea, 2017-05-23:
Is there a way to indicate a GattCharacteristic? From what i see you can read if a characteristic in Indictable but is there a way to indicate it?

Cierpliwy, 2017-05-23:
You can monitor characteristic's value by enabling notifications (without ACK) or indications (with ACK). On iOS there is only one function to do it: setNotifyValue() which prefers notifications over indications.

As stated in documentation:
https://polidea.github.io/react-native-ble-plx/class/react-native-ble-plx/src/BleManager.js~BleManager.html#instance-method-monitorCharacteristicForDevice

"If notifications are enabled they will be used in favour of indications."

So only option to use indications in platform independent way is to disable notifications in your device's firmware and enable indications only. I hope it clarifies things.

alexandrapersea, 2017-05-25:
Ok so what I need to do is indicate a characteristic and then write a command to it. My device responds with either a value or an error.
My problem is that the device does nothing, and i don't get any response back. I don't know exactly what I'm doing wrong.

My code looks something like this:
```
// Command comes from server so im 100% sure its correct.
cmd = "1020c96d8d2ca889d16e0035876424c7b95ee102423aba891e9e66068640abbf252803"
// my device accepts only packages of 20 bytes so i split it into 2
cmd_1 = "1020c96d8d2ca889d16e0035876424c7b95ee1"
cmd_2 = "02423aba891e9e66068640abbf252803"
// i encode it into base64
first_message = "MTAyMGM5NmQ4ZDJjYTg4OWQxNmUwMDM1ODc2NDI0YzdiOTVlZTE="
second_message = "MDI0MjNhYmE4OTFlOWU2NjA2ODY0MGFiYmYyNTI4MDM="

device.monitorCharacteristicForService('079593bb-1785-41c9-ae9c-3214c30428dc',
                                                                   'c9003789-b61c-47f0-b22b-a526d06c1a74', 
                                                                   (error, characteristic) => {
  if (error) {
    console.log(error);
   }
   else {
     console.log(characteristic.value)
    }
})

device.writeCharacteristicWithResponseForService('079593bb-1785-41c9-ae9c-3214c30428dc',
                                                                                       'c9003789-b61c-47f0-b22b-a526d06c1a74',
                                                                                       'first_message')
  .then(resp => {
     // After first package was sent, send the second one
      device.writeCharacteristicWithResponseForService('079593bb-1785-41c9-ae9c-3214c30428dc',
                                                                                           'c9003789-b61c-47f0-b22b-a526d06c1a74',
                                                                                           'second_message')
        .then(resp => console.log(resp))
        .catch(err  => console.log(err))
  })
  .catch(err  => console.log(err))
```

closed 2017-05-30 by alexandrapersea

bhaveshj-cuelogic, 2019-02-26:
how do I achieve setNotifyValue:YES forCharacteristic.
please suggest some example for that