dotintent/react-native-ble-plx #79: framework not found BleClientManager for architecture arm64

microwavesafe, 2017-05-26:
I started by trying to install it automatically.

```npm install react-native-ble-plx```
```react-native link```

which works, but does through up some warnings when running the build script. The warning seems to suggest it can't open Rx.playground in workspace Rx.xcworkspace. Although I have verified that at least the files exist, and the build never fails.

I then tried to follow the manual steps. I noticed that ```Always Embed Swift Standard Libraries``` was set to no and it didn't have the Run Script Phases, so I added those, rebuilt the project but same outcome.

I'm more familiar with Android so I'm struggling to debug this. Is there something obvious I'm missing?

Cierpliwy, 2017-05-26:
There is a note in documentation that regardless of 
```
npm install react-native-ble-plx
react-native link
```
you have to continue manual installation setup from step 7.
Did you do that?

microwavesafe, 2017-05-26:
I knew it would be simple when I could see it! I thought I had, but it looks like I actually didn't do 7.

closed 2017-05-26 by microwavesafe