dotintent/react-native-ble-plx #80: undefined is not an object (evaluating this.manager.startDeviceScan())

watadarkstar, 2017-05-26:
[image]

I have this problem that only occurs when I navigate away from the component and the BleManager gets destroyed and deleted then navigate back the component gets mounted again I recreate the BleManager and the scan fails. Even if I put the manager in a redux store and don't recreate it the manager, the scan gives me the same error....

My code is:

```javascript
componentWillMount() {
        this.manager = new BleManager();
        this.manager.onStateChange((state) => {
            this.setState({ ...this.state, devices: {}, bluetoothState: state });
            if (state === 'PoweredOn') {
                this.scan();
            }
        }, true);
    }

scan() {
        this.manager.startDeviceScan(null, null, (error, device) => {
            if (error) {
                // Handle error (scanning will be stopped automatically)
                return;
            }
            if (device.name === null) {
                return;
            }

            if (_.includes(device.name.toLowerCase(), 'penta')) {
                this.manager.stopDeviceScan();
            } else {
                return;
            }

            this.setState({
                ...this.state,
                devices: { ...this.state.devices, [device.name]: device }
            });
        });
    }

componentWillUnmount() {
        this.manager.stopDeviceScan();
        this.manager.destroy();
        delete this.manager;
    }
```

watadarkstar, 2017-05-27:
I fixed the error by doing `const manager = this.manager = new BleManager()` and then passing `manager` into `scan(manager)` but the scan still doesn't return any devices when I come back to the scan screen and re-mount the component again. Any ideas?

Cierpliwy, 2017-06-02:
I will try to avoid calling destory() in unmount function right now. There are few things which may lead to unexpected behaviour.

1). Destroying is very safe from native module perspective.
2). If there are any not finished action on BLE side they will be cancelled afterwards.
3). There is possibility that native module will emit events just before it was destroyed.
4). Events may arrive after component was unmounted leading to bas behaviour.

I need to think how can I tackle that and if it is even possible to do without inconvenience.

watadarkstar, 2017-06-05:
@Cierpliwy So are you suggesting that I move `this.manager` into a redux store and maybe that will solve my issue of `startDeviceScan` not working when the user returns to the screen after leaving?

Cierpliwy, 2017-06-06:
I'm looking into that as I want to solve any issues around destruction as soon as possible. For now you can move manager to a place which will not be destroyed (outside of component or to a component which lives for whole app duration)

Cierpliwy, 2017-06-07:
@watadarkstar I implemented few fixes for proper destruction of all promises and subscriptions which BleModule calls .destroy() function. You can check it with following branch:

```
"react-native-ble-plx": "git://github.com/Polidea/react-native-ble-plx.git#fix_destory_function"
```

watadarkstar, 2017-06-07:
@Cierpliwy Awesome I will take a look at it this week! Thank you :)

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy