dotintent/react-native-ble-plx #82: Xcode 8.2 no problem,but in Xcode 8.3.2  yld: lazy symbol binding failed: Symbol not found: __TMaCs29_NativeDictionaryStorageOwner

Muriel0603, 2017-06-01:
Referenced from: /Users/xxx/Library/Developer/CoreSimulator/Devices/532F245F-3273-49C4-A3F2-CA22D8E2118C/data/Containers/Bundle/Application/F5331AD8-DFDD-4FF3-AFE0-50A343A798E2/xxx.app/Frameworks/BleClientManager.framework/BleClientManager
  Expected in: /Users/xxx/Library/Developer/CoreSimulator/Devices/532F245F-3273-49C4-A3F2-CA22D8E2118C/data/Containers/Bundle/Application/F5331AD8-DFDD-4FF3-AFE0-50A343A798E2/xxx.app/Frameworks/libswiftCore.dylib

Cierpliwy, 2017-06-02:
I just checked and library is working properly on XCode 8.3.2. Make sure to clear all caches and recompile all frameworks.

jiazil, 2017-07-21:
my here is happen too ,my xcode is 8.3.3

Referenced from: /Users/xxx/Library/Developer/CoreSimulator/Devices/4962FCDD-6373-4712-B8BE-2A5E8A6E309F/data/Containers/Bundle/Application/D91AE773-9135-4358-9EB5-6E6508143A95/leisure.app/Frameworks/BleClientManager.framework/BleClientManager
  Expected in: /Users/xxx/Library/Developer/CoreSimulator/Devices/4962FCDD-6373-4712-B8BE-2A5E8A6E309F/data/Containers/Bundle/Application/D91AE773-9135-4358-9EB5-6E6508143A95/xxx.app/Frameworks/libswiftCore.dylib

revolucion, 2017-09-21:
I had the same issue with Xcode 8.3.3.

I ended up copying the /node_modules/react-native-ble-plx folder from the /examples/ReactBLEScanner/node_modules folder, and replacing it in my own project instead of running the npm install command.

This worked, for some reason.

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy