dotintent/react-native-ble-plx #83: couldn't detect near by bluetooth device

biks152207, 2017-06-05:
Hi i couldn't get nearby device even the bluetooth is turn on in other device.
`  constructor(props) {
    super(props);
    this.state = { text: ''};
    this.manager = new BleManager();
    this.scan = this.scan.bind(this);
    this.error = this.error.bind(this);
    this.scanAndConnect = this.scanAndConnect.bind(this);
  }

  scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
      console.log(error);
      console.log(device);
    })
  }

  componentWillMount() {
      const subscription = this.manager.onStateChange((state) => {
          if (state === 'PoweredOn') {
              this.scanAndConnect();
              subscription.remove();
          }
      }, true);
  }`

Scan and connect should provide me the nearby device.but doesn't show anything..i tried multiple times.no luck.

michalkow, 2017-06-05:
Why do you wrap `this.scanAndConnect();` in `onStateChange`?
Is the idea that scan should start when you turn on your Bluetooth? Cold you just add onClick trigger for the scan, make sure it works? Because `scanAndConnect` looks good and should return console logs.

biks152207, 2017-06-05:
@michalkow  i tried scanning device on onClick also.i haven't added the view part here..but i have called scanAndConnect with onPress

michalkow, 2017-06-05:
It should work then I think... Are you using it for Android or iOS? Did you set up all Bluetooth permissions?
Did you successfully run react-native-ble-plx example app Sniffator?

biks152207, 2017-06-05:
i am not sure about the Sniffator..but i turned on my bluetooth other bluetooth device and my test device connected to my computer through usb and its bluetooth is also turned on..and started scanning the device..but it couldn't find my other device whose bluetooth is on.

michalkow, 2017-06-05:
Try to download and use Sniffator so you can be sure it is problem with your code, and not with he device

biks152207, 2017-06-05:
what if sniffator is able to scan the devices..i have been trying to resolve this issue for 3-4 days..no luck at all.

Cierpliwy, 2017-06-05:
Please send me a minimal javascript file which is responsible for scanning. I'll test it.

biks152207, 2017-06-05:
Hi @Cierpliwy below is the code..i have reduced my code

`import React, { Component } from 'react';
import {BleManager} from 'react-native-ble-plx';

class Settings extends Component {  // eslint-disable-line

  constructor(props) {
    super(props);
    this.manager = new BleManager();
    this.scanAndConnect = this.scanAndConnect.bind(this);
  }

  scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
      console.log(error);
      console.log(device);
    })
  }

  componentWillMount() {
      const subscription = this.manager.onStateChange((state) => {
          if (state === 'PoweredOn') {
              this.scanAndConnect();
              subscription.remove();
          }
      }, true);
  }

  render() {
    return (
        <View>
          <View>
            <Text>Scan nearby device</Text>
          </View>

          <View >
            <Button style={styles.scannerButton} onPress={this.scan}>
                <Icon name='bluetooth' style={{color: 'white', fontSize: 20, fontWeight: '800'}}/>
                <Text style={{color: 'white', fontSize: 20}}>Scan</Text>
                <Spinner color="#fff"/>
            </Button>
          </View>
        </View>
    );
  }
}

export default connect(null, null)(Settings);
`

biks152207, 2017-06-05:
@Cierpliwy @michalkow  i tried downloading Sniffator and started scanning..but it doesn't find bluetooth device..the other device is ios..I suspect that it couldn't find ios bluetooth device?

michalkow, 2017-06-05:
I am not sure what you are trying to do, but it sounds like this: 
https://stackoverflow.com/questions/18884705/transfer-data-between-ios-and-android-via-bluetooth

Cierpliwy, 2017-06-05:
I have simplified your code a bit:
```js
// @flow

import React, { Component } from "react";
import { Button, View, Text } from "react-native";
import { BleManager } from "react-native-ble-plx";

export default class TestComponent extends Component {
  constructor(props) {
    super(props);
    this.manager = new BleManager();
    this.scanAndConnect = this.scanAndConnect.bind(this);
  }

  scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
      console.log(error);
      console.log(device);
    });
  }

  componentWillMount() {
    const subscription = this.manager.onStateChange(state => {
      if (state === "PoweredOn") {
        this.scanAndConnect();
        subscription.remove();
      }
    }, true);
  }

  render() {
    return (
      <View style={{flex: 1}}>
        <Button onPress={this.scanAndConnect} title={"CLICK!"}>
          <Text style={{ color: "white", fontSize: 20 }}>Scan</Text>
        </Button>
      </View>
    );
  }
}
```

It works fine on Android. I need to wait for my iPod to be charged before I can verify it on iOS.

biks152207, 2017-06-05:
@Cierpliwy so that means it works in andorid to android.i was checking from android to ios.Please verify it and let me know..very much thanks

Cierpliwy, 2017-06-05:
On iOS it works fine. Please add log level to your code:
```js
import { BleManager, LogLevel } from "react-native-ble-plx";
...
this.manager.setLogLevel(LogLevel.Verbose)
...
```
and check logs in XCode/Android Studio.  On iOS it looks like this:
```
[RxBLEKit|DEBG|16:44:07.452]: CentralManager(401912704) didDiscover(peripheral: Peripheral(uuid: EA22F5F6-7EB0-0C8B-57DB-855A6650AEAE, name: Optional("BLE Name")), rssi: -94)
[RxBLEKit|DEBG|16:44:07.469]: CentralManager(401912704) didDiscover(peripheral: Peripheral(uuid: 21BAC7C4-34EE-AA5B-FFD5-AD99CCB05FAF, name: nil), rssi: -90)
[RxBLEKit|DEBG|16:44:07.471]: CentralManager(401912704) didDiscover(peripheral: Peripheral(uuid: 5450896F-2654-34AC-ACED-E485FF307422, name: nil), rssi: -82)
```

biks152207, 2017-06-05:
@Cierpliwy thank you very much..let me try.

biks152207, 2017-06-05:
@Cierpliwy  Could you pls try scanning from android to search ios ble..

biks152207, 2017-06-06:
@Cierpliwy Have you checked the scanning between android and ios?

Cierpliwy, 2017-06-06:
What do you mean by "scanning between android and ios"? Did you create iOS application which is advertising in peripheral mode and want to scan it with Android?

biks152207, 2017-06-06:
@Cierpliwy  can scanning through android device find the ios ble(uuid)?

Cierpliwy, 2017-06-06:
Well you can't normally scan iOS device. You have to have specific application which will advertise in Peripheral mode to do so.

biks152207, 2017-06-07:
@Cierpliwy  what actually advertise implies to?i am not sure about it.could you pls explain this advertising peripheral mode?

Cierpliwy, 2017-06-07:
Devices using BLE work in one of two modes: Central (which this library allows) and Peripheral. In most of use cases applications on your phone implement Central mode which allows them to properly scan, connect to and communicate with devices. This is normal flow.

There is also possibility to create and _application_ in Peripheral mode on iOS and newest Androids. This is not supported by this library, your iPhone/Android by itself won't start working as Peripheral. You need to create/install application which does just that. When it is running, it advertises itself as a specific BLE device and may allow one Central to connect to it.

I recommend that your read more about how BLE works.

sikhote, 2017-07-10:
@Cierpliwy I have an issue where I can see the device within my Android Bluetooth settings and I can pair with it successfully there and use it in a different app. Once I start the React Native app it can see lots of other Bluetooth devices, but not the device that is powered up next to the computer. I have tried searching for the device while it is paired, when it is not, and doing a search while the device is in pairing mode. Have you experienced this before? I suspect it is the device I need to pair with causing issue, but hoping for another path to explore.

Thanks for any thoughts!

closed 2019-03-23 by stale[bot]

tal987, 2020-03-17:
@Cierpliwy hey I just want to make sure... with this library can I discover other mobile devices like other iOS/android phones?

DmitryB11, 2021-05-12:
> @Cierpliwy I have an issue where I can see the device within my Android Bluetooth settings and I can pair with it successfully there and use it in a different app. Once I start the React Native app [...]

Hi, I have the same problem. Did you solve it?

taj567, 2021-11-26:
const scanDevices = async () => {
    const btState = await manager.state()
    console.log(btState,"btstate scandevices")
    if (btState!=="PoweredOn") {
      alert("Bluetooth is not powered on");
      return (false);
    }
    const permission = await requestPermission();
    console.log(permission,"permission")
    if (permission) {
      manager.startDeviceScan(null, null, async (error, device) => {
          if (error) {
            console.log(error);
            return
          }
          if (device) {
            console.log(device)
            // console.log(`${device.name} (${device.id})}`);
            const newScannedDevices = scannedDevices;
            newScannedDevices[device.id] = device;
            await setDeviceCount(Object.keys(newScannedDevices).length);
            await setScannedDevices(scannedDevices);
          }
      })
    }
  }

using react-native-ble-plx not detecting all nearby devices ?

SyntaxError843, 2024-01-18:
For the people that can't detect iBeacon/Eddystone BLEs try editing your android manifest

`<manifest xmlns:android="http://schemas.android.com/apk/res/android" xmlns:tools="http://schemas.android.com/tools">`
`<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags"/>`