dotintent/react-native-ble-plx #84: Read or monitor characteristic

jchirschy, 2017-06-06:
Hello,

It may not be an issue, but maybe a misunderstanding of how to use this API properly.
I'm trying to use UART over BLE with an Adafruit Bluefruit LE UART Friend.
Scanning, connecting and writing are all working well.

RX characteristic is supposed to be readable, but it is reported as false. Sniffator app is reporting it as well. So I setup notification following the code provided here : 
https://www.polidea.com/blog/ReactNative_and_Bluetooth_to_An_Other_level/

With little data, reading directly the value of the RX characteristic works well, but in case of long text (> 20 bytes), it is broken or some parts are repeated or missing.
Using the provided Bluefruit app, UART works well including very long texts, meaning Adafruit Bluefruit LE UART Friend works well.

Could you please explain me a little bit more the following points :
Is there an known issue with isReadable ?
How could you manage long text with notifications ?
Is there a way to setup bauds rate ? Or is it managed at a lower level ?

Thanks for your help

Cierpliwy, 2017-06-07:
Hello,

> Is there an known issue with isReadable ?

It might be a device firmware bug or an issue in our library. I have for example a TI peripheral which passes wrong data in characteristic descriptor. The best way is to do an operation and see if it works. Also please check if nRF connect application is displaying same information.

> How could you manage long text with notifications ?

According to BLE Specification Vol 3 Part F 3.4.7.1, max notification size is ATT_MTU - 3. Which means it can range from 20 to 519 bytes of pure data depending on negotiated MTU value between devices. On iOS for example you can't negotiate value by yourself.

Also you need keep in mind that:
> If the attribute value is longer than (ATT_MTU-3) octets, then only the first (ATT_MTU-3) octets of this attributes value can be sent in a notification.

So that's why your data may look corrupted or missing.

> Is there a way to setup bauds rate ? Or is it managed at a lower level ?

It's not a job of BLE transport layer. I think you should check documentation of your peripheral device if it is configurable etc.

Is your Bluefruit app running on iOS or Android?

jchirschy, 2017-06-07:
Thanks for your quick response.

When I try to read the RX characteristic that is described as not readable, I got the following exception : 
> BleGattCannotStartException{macAddress=C2:0A:F6:BC:B6:44, bleGattOperationType=BleGattOperation{description='CHARACTERISTIC_READ'}}
It seems to confirm that the characteristic is effectively not readable ?

I realize, I have not a good understanding of BLE.
I was thinking that size was limited to only 20 bytes.
To write from the central, I slice text in chunks of 20 bytes.
Is it really necessary ?

What I found out by setting up notification, is I can get 2 or more notifications even if the payload is less than 20 bytes. The asynchronous way of receiving data may exchange the order of 'packets' ?

Have you got a complete source for dealing UART over BLE with this library, from the central point of view ?

Thanks for your help.

Cierpliwy, 2017-06-07:
> It seems to confirm that the characteristic is effectively not readable ?

Could be. You should also get error status, which is telling you a reason of failed operation. In case of this exception: `BleGattCharacteristicException{macAddress=B4:99:4C:34:DE:77, status=3` it
would be `3` which is `GATT_WRITE_NOT_PERMIT `.

> Is it really necessary ?

When MTU is your limit, but your device supports long reads and long writes I assume that iOS and Android should handle chunking your data automatically. Of course it is not working for notifications which are always limited by MTU and you need to read value manually to get all data.

>  The asynchronous way of receiving data may exchange the order of 'packets' ?

I doubt it when you are talking about receiving data from the same characteristic.

> Have you got a complete source for dealing UART over BLE with this library, from the central point of view ?

I've never done that, so I can't help.

jchirschy, 2017-06-07:
I didn't get any status with the Exception thrown.
Where do you get this information ?

I try another lib that handle BLE.
RX characteristic appears not to be readable as well.
With this one, reading value on notifications works well even with long text.
Is there an issue with your library ?

Cierpliwy, 2017-06-07:
What do you mean by reading value on notifications? You are suggesting that notifications are sending larger data chunks by using another library?

jchirschy, 2017-06-08:
Hi,

Using another library for using BLE with React-Native, I received as many notifications as necessary.  On the emitted event, I got values that I just needed to concatenate to emulate an UART.
I tried to do the same thing with react-native-ble-plx, but with no success. It seems that some notifications are missing or sent twice...

Cierpliwy, 2017-06-08:
Hello,

Could you tell me which library is it? I would love to check how they are handling this case in their native module.

jchirschy, 2017-06-08:
Yes, of course :
https://github.com/innoveit/react-native-ble-manager

Cierpliwy, 2017-06-08:
I would really appreciate if you could check following branch:
```
"react-native-ble-plx": "git://github.com/Polidea/react-native-ble-plx.git#ver_0_6"
```
And if it fixes your issues.

jchirschy, 2017-06-08:
Hi Cierpliwy,

It works very well now.
Thanks.

Cierpliwy, 2017-06-08:
Glad to hear that!

closed 2017-06-08 by Cierpliwy

jchirschy, 2017-06-28:
Hello @Cierpliwy,

It seems that last Master branch reproduces the same issue.
When I come back to git://github.com/Polidea/react-native-ble-plx.git#ver_0_6, everything is ok.

jchirschy, 2017-06-28:
My bad, seems using wrong package.json with bad semver.
Sorry for useless disturbance.

jchirschy, 2017-10-25:
Hello @Cierpliwy,

The app in development is nearly to a first milestones thanks to this wonderful library.
I'm testing a lot those days and I'm facing *sometimes* some troubles in receiving notifications.

Here is the description of the use case : 
1. Using peripheral to send a rather big json file, the same one each time
1. Setting notification on the central side and receiving a lot of packets.
1. Upon receiving `\n`, concatenate those packets and checking the `CRC` sent by the peripheral
1. Approximately one time out of 6, the central does not calculate the same `CRC`, meaning a diff between what was sent by the peripheral. (By the way, since the json that is sent is the same for each test, the `CRC` sent by the peripheral is always the same).
1. Having analysed the received json, 'some' characters are missing. A BLE spy confirms that the json that was sent has no missing characters. This issue does not seem to appear with a little json.

Is there possibly a loss of some packets sometimes ? that could happen more often with a lot of packets ?
What could be a way to get some deeper log at a lower level on a Android central ?
Is there some tweaking that could be done to the react native library, the native one or the central itself ?
Some helpful advices would be appreciate.

Thanks in advance.
Sincerily
JeanClaude

Cierpliwy, 2017-10-26:
Hello,

Thank you for testing this library and sharing your problems with us. It's the only way we can improve it. I would definitely start by analysing how native dependencies are handling notifications. You can enable native libraries logging by calling `manager.setLogLevel(LogLevel.Verbose)`. They will be visible on XCode/Android Studio.

If your test requires big files to properly reproduce an issue I would suggest to generate json with predictable pattern and check if every notification is following it. If you are working on Android on branch 0.6.4 you can additionally edit native code at:

**./node_modules/react-native-ble-plx/android/src/main/java/com/polidea/reactnativeble/BleModule.java:881**

which contains method  `safeMonitorCharacteristicForDevice(final Characteristic characteristic...` and from line 934 to 940:

```
@Override
public void onNext(byte[] bytes) {
    characteristic.logValue("Notification from", bytes);
    WritableArray jsResult = Arguments.createArray();
    jsResult.pushNull();
    jsResult.pushMap(characteristic.toJSObject(bytes));
    jsResult.pushString(transactionId);
    sendEvent(Event.ReadEvent, jsResult);
}
```
you can access `bytes` variable with received data from notification. I would suggest to monitor it and check if next emissions are following pattern which you will be sending from device. You can debug/modify this file directly and use  RxBleLog.v("your message") for logging - just ensure that your Android app will be rebuild after that.

If there will be an issue here then I would suggest to sniff BLE packets or we can dig further. If not, we can search for problems on JS side of things.

Best regards,
Przemek

jchirschy, 2017-10-31:
Hello,

Thanks for all of these informations.

I did not have enough time to investigate before the delivery of an alpha version. For the moment, I have handled the possible issue with repeating the transmission.  As soon as I can, I will try to get some deeper log following your advice.

By the way, could you tell me what rate of transmission do you succeed to get with this library ?
Have you also noticed an asymmetry in "bandwith" between reception and emission ?
Is there a necessity to chunk data in packet of 20 bytes before sending out ?
Is there a roadmap to bridge the `base64` encoding and decoding process as it is stated earlier in this thread ?

Thanks in advance for your precious help.
JeanClaude

LingboTang, 2017-12-15:
@jchirschy @Cierpliwy Is the UArt Style supported now? I would like to use this package for Uart Firmwares.

jchirschy, 2017-12-15:
I use this library in an UART context and it works very well.
You will have to know which characteristics to use as RX and TX ; it depends on the peripheral. For that, you could use `nRF Connect` for Android or `Lightblue` with IOS to find them out. And then, you just use notification and writing mechanism.

stevemk14ebr, 2018-05-17:
For those who follow me i had the same issue with the bluefruit module. Here is working code, CCCD notifications are enabled by monitorCharacteristic method now:

//page 105 https://cdn-learn.adafruit.com/downloads/pdf/introducing-the-adafruit-bluefruit-spi-breakout.pdf

```
this.UART_SERVICE = "6e400001-b5a3-f393-e0a9-e50e24dcca9e";
this.RX_CHARACT = "6e400003-b5a3-f393-e0a9-e50e24dcca9e";
this.TX_CHARACT = "6e400002-b5a3-f393-e0a9-e50e24dcca9e";
connectedDevice = await connectedDevice.discoverAllServicesAndCharacteristics();

    // Write to TX
    connectedDevice.writeCharacteristicWithResponseForService(
      this.UART_SERVICE,
      this.TX_CHARACT,
      "aGVsbG8gZnJvbSBqcw=="
    );

    // monitor RX, CCCD notifications enabled automatically by lib
    let char = await connectedDevice.monitorCharacteristicForService(
      this.UART_SERVICE
      this.RX_CHARACT,
      (error, chr) => {
        console.log('Received:', chr.value);
      }
    );

```

alexbuicescu, 2020-01-31:
> Hello @Cierpliwy,
> 
> The app in development is nearly to a first milestones thanks to this wonderful library.
> I'm testing a lot those days and I'm facing _sometimes_ some troubles in receiving notifications.
> 
> Here is the description of the use case :
> 
> 1. Using peripheral to send a rather big json file, the same one each time
> 2. Setting notification on the central side and receiving a lot of packets.
> 3. Upon receiving `\n`, concatenate those packets and checking the `CRC` sent by the peripheral
> 4. Approximately one time out of 6, the central does not calculate the same `CRC`, meaning a diff between what was sent by the peripheral. (By the way, since the json that is sent is the same for each test, the `CRC` sent by the peripheral is always the same).
> 5. Having analysed the received json, 'some' characters are missing. A BLE spy confirms that the json that was sent has no missing characters. This issue does not seem to appear with a little json.
> 
> Is there possibly a loss of some packets sometimes ? that could happen more often with a lot of packets ?
> What could be a way to get some deeper log at a lower level on a Android central ?
> Is there some tweaking that could be done to the react native library, the native one or the central itself ?
> Some helpful advices would be appreciate.
> 
> Thanks in advance.
> Sincerily
> JeanClaude

Hi @jchirschy 
We see the same behaviour. Did you find any solution beside resending the offending message?

jchirschy, 2020-03-11:
Hi @alexbuicescu,

No, not at this time.
Resending is enough for my purpose.

alexbuicescu, 2020-03-11:
Thank you @jchirschy. In case anyone comes across this, here's what we did and works for us:
1) we use UART (not a choice, our device's hardware supports only UART)
2) calculate and append the message's CRC at the start of the message
2) we split our message in 20 bytes chunks (basically doing a manual mtu). As @jchirschy mentioned this bug does not happen with smaller messages (little json) and that's why we split the message manually
3) send all the message's chunks with a sleep of 32ms between them (you can play with this until you find just the right value for the majority of your cases, 32ms is what worked best for us)
4) in case the CRC does not match, tell the transmitter to repeat step 3) but with a sleep of 100ms between chunks

Almost all the time step 4) solves the issue. In cases where it didn't, there other problems causing this like receiver and transmitter too far away from each other, objects in the way interfering with the bluetooth waves, ble connection unstable, etc. But even in these cases we had lots of success.

Also, for this to work, we came up with a protocol so we know when we received the full message and start to create the next message. More specifically, we used a start byte (something like "<") and stop byte (something like ">") and when we found the start byte we would concatenate the bluetooth notifications until we received the stop byte.

ShivBagherwal, 2021-03-11:
> Thanks for your quick response.
>
> When I try to read the RX characteristic that is described as not readable, I got the following exception :
>
>
> It seems to confirm that the characteristic is [...]

@jchirschy please increase the MTU size (https://github.com/Polidea/react-native-ble-plx/wiki/MTU-Negotiation)