dotintent/react-native-ble-plx #87: Android BleGattException status 129

1453026, 2017-06-13:
Hi,
I'm recently designing the Ble devices for detecting beacons. I have some problem with the function when discovering the characteristics of the beacons.
This is the problem: ERROR: BleGattException(macAddress=24:0A:C4:07:03:C2, status=129 (0x81 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h),bleGattOperationType=BleGattOperation{description='SERVICE_DISCOVERY'}
I found that your app Sniffator also has the same problem discovering to the beacons. And this problem occurs with Android only, it works fine in iOS.
Does anyone know how to fix this ?
Thank you very much.

Cierpliwy, 2017-06-13:
I'm afraid I won't be able to help with this issue. I was talking with my colleague and he suggested to restart your BLE devices (phone and beacon). It is `GATT_INTERNAL_ERROR` so it could be an implementation error on Android or firmware side.

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy