dotintent/react-native-ble-plx #88: write characteristic get BleGattException status= 19

kinber95, 2017-06-13:
On some off my android device I get this error when I want to write a characteristic for a service:

`Error: BleGattException{status=19, bleGattOperation=BleGattOperation{description='CONNECTION_STATE'}}
    at createErrorFromErrorData (NativeModules.js:115)
    at NativeModules.js:78
    at MessageQueue.__invokeCallback (MessageQueue.js:296)
    at MessageQueue.js:120
    at MessageQueue.__guard (MessageQueue.js:218)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (MessageQueue.js:119)
    at debuggerWorker.js:71`

On other Android device there are no problems

Cierpliwy, 2017-06-13:
This is an `GATT_RSP_EXEC_WRITE` error which might suggest that your device is not sending response to Android in time. Please check if this behaviour also occurs in nRF Connect application.

kinber95, 2017-06-13:
@Cierpliwy  no in nRF Connect I can send my characteristic value to my ble device.

Should I use writeCharateristicWithoutResponse() ?

DohaBasem, 2018-04-18:
@kinber95 did you find a solution to this issue or got any source that could elaborate more?

kinber95, 2018-04-18:
@DohaBasem use writeCharacteristicWithoutResponse() and when you get the 19 error retry the write command.

BLE on Android sucks, especially on Motorola or Sony devices. We don‘t use rn for Android any more. Go back to the roots and use Java. Its sucks also  but  less than rn.

Cheers

closed 2019-03-23 by stale[bot]